chipsalliance/verible #1647: Go to definition does not work if there is no file list

hzeller, 2023-01-27:
- IDE/Editor; version  ? vscode, 1.74.3
 - What other SystemVerilog plugins are active alongside ? None.
-----

**What activity failed**
go-to-definition when there is no file list.

Simple project: one SystemVerilog file, but no `verible.filelist`
Attempt to resolve a symbol in the same file.

**Expectation**
That the file is analyzed and the symbol found.

**What actually happened**
On 'go-to-definition', the editor says that the definition was not found, even though it is in the file.

**Test case**

Here is an example file. It should be able to resolve `baz` as it is in the same file, but fails.

```systemverilog
module baz();
endmodule

module foo();
  bar x;
  baz dd;
  flop xy;
endmodule
```

**Logfiles**

The logfile indicates that the symbol table attempts to find the file list and then gives up. Expectation would be that it then at least attempts to analyze the local file the editor has open; then it would be able to resolve the `baz` symbol definition.

```txt
I0127 12:20:28.895740  219540 symbol-table-handler.cc:96] LoadProjectFileList
W0127 12:20:28.895836  219540 symbol-table-handler.cc:48] Could not find verible.filelist file in the project root (/home/hzeller/tt/.):  NOT_FOUND: No matching file found.
```

hzeller, 2023-01-27:
It does work when an empty file-list exists (so a `touch verible.filelist`).
That indicates that we could potentially just continue with a `verible::FileList` that is empty.

glatosinski, 2023-01-30:
I created an initial PR in https://github.com/chipsalliance/verible/pull/1652 resolving the issue when we want to jump to symbol defined in the same file.

It will also work for jumps between edited buffers.

@hzeller - should we also automatically load files e.g. from the current workspace directory, or should we require `verible.filelist` to be provided to open files for symbols?

hzeller, 2023-01-30:
Opening  files in the root might be something to consider in the future, but we need to probably figure out some details first, e.g. only open files in the current directory (could not be enough) or recursively load all the files (could take forever and might not want is needed)..

For now, I think #1652 should be sufficient.

closed 2023-01-30 by hzeller