chipsalliance/verible #1651: Publish verible.vsix to VSCode marketplace

hzeller, 2023-01-28:
The Verible Language server extension is a small vsix file that needs to be installed in VSCode, for which we provide [installation instructions](https://github.com/chipsalliance/verible/tree/master/verilog/tools/ls#vscode).

We should explore if we can provide that in the VSCode marketplace
  https://marketplace.visualstudio.com/VSCode

Here are details of what needs to be done. I have domains for verible (that are about the be transferred to chipsalliance), we need to set them up for documentation and make sure to run the DNS so that we can verify our domain:
https://code.visualstudio.com/api/working-with-extensions/publishing-extension#advanced-usage

The nice thing is, that this only really needs to be installed once and rarely updated as this is only the interface to the editor while actual functionality comes in the binary `verible-verilog-ls` that is updated with a new binary release.

hzeller, 2023-02-14:
Assigning to Peter to triage and sequence needed steps. We should start figuring out making a Verible site and DNS so that we can verify against the marketplace.

pgielda, 2023-02-14:
Seems that to do that we need to create a "Publisher". It could be "CHIPS Alliance", what do you think about that? I can discuss within the organization on how to get this verification through for the https://chipsalliance.org/ website... But obviously it might be that we would need to publish via this domain (I wonder if we could do it via a subdomain, e.g. https://verible.chipsalliance.org/) or we can try to register "Verible" publisher -- which is maybe a little bit weird as then you have to also provide things like "Company website, etc. But maybe we could add verible.org as the "Domain name" and CHIPS Alliance as the "Company Website".

I am confused if this is just for this "blue tick as seen here: https://code.visualstudio.com/api/working-with-extensions/publishing-extension#verify-a-publisher
or if this is where you later actually put the packages. If its just for the tick I would go with CHIPS Alliance as the publisher...

Will ask someone to verify this.

hzeller, 2023-04-28:
The headline in https://marketplace.visualstudio.com/items?itemName=CHIPSAlliance.verible currently says

## Verible Language Server

Maybe it would be also good to include SystemVerilog in that, as 'Verible' might be unknown

## Verible SystemVerilog Language Server (Lint, Format, go-to-definition,...)

hzeller, 2023-04-28:
There are currently other plug-ins that use Verible, and all of these have SystemVerilog in their headline:

https://marketplace.visualstudio.com/search?term=verible&target=VSCode&category=All%20categories&sortBy=Relevance

They have collected quite some downloads (and have been around for arguably longer). But adding SystemVerilog in the headline can probably improve our marketing there.

hzeller, 2023-05-05:
Alright, changed the title to contain `SystemVerilog`
[image]

I think we can mark this as completed.

closed 2023-05-05 by hzeller