chipsalliance/verible #1658: Build failed on ubuntu 22.04

LoveMyPillow, 2023-01-31:
My Operation System is ubuntu 22.04 and I'd like to build the Verible Project, and I use nix package manager to solve the dependency. However, on nix-shell I use the command "bazel build -c opt //..." as the readme file said, and I meet a fatal error on git. 

It seems that there is a python file named "build-version.py" and it runs git log command. However, there doesn't exist a git repo on verilbe root directory, as a result, git meets a fatal error. So my question is that should I use git to create a repo here and build it again? Thanks!

And error message is attached.
//===========================ERROR MESSAGE=====================================
[nix-shell:~/S-Verible/verible-0.0-2776-gbaf0efe9]$ bazel build -c opt //...
WARNING: Download from https://mirror.bazel.build/github.com/westes/flex/releases/download/v2.6.4/flex-2.6.4.tar.gz failed: class com.google.devtools.build.lib.bazel.repository.downloader.UnrecoverableHttpException GET returned 404 Not Found
INFO: Analyzed 1891 targets (0 packages loaded, 0 targets configured).
INFO: Found 1891 targets...
ERROR: <builtin>: BazelWorkspaceStatusAction stable-status.txt failed: Failed to determine workspace status: Process exited with status 1
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "/home/hiramhsu/S-Verible/verible-0.0-2776-gbaf0efe9/bazel/build-version.py", line 52, in <module>
    main()
  File "/home/hiramhsu/S-Verible/verible-0.0-2776-gbaf0efe9/bazel/build-version.py", line 42, in main
    date = os.environ["GIT_DATE"]
  File "/nix/store/abax98471z8fshv4b9p46bkh3lxmpy0z-python3-3.10.9/lib/python3.10/os.py", line 680, in __getitem__
    raise KeyError(key) from None
KeyError: 'GIT_DATE'
INFO: Elapsed time: 0.444s, Critical Path: 0.10s
INFO: 9 processes: 9 internal.
FAILED: Build did NOT complete successfully

hzeller, 2023-01-31:
It looks like you did not check out the version from git, so the script that attempts to extract that from the repo does not work.
We should probably make that script a bit more robust to fall back to some generic version string when it can't work.

In the meantime I suggest to clone the repository from git, so that the `.git/` directory is available.

```
git clone https://github.com/chipsalliance/verible.git
```

hzeller, 2023-01-31:
You can also set the environment variable `GIT_DATE` and `GIT_VERSION` to some values.

(these are values that then show up in the when invoking the binaries with `--version`, e.g. `verible-verilog-lint --version`)

LoveMyPillow, 2023-01-31:
> It looks like you did not check out the version from git, so the script that attempts to extract that from the repo does not work. We should probably make that script a bit more robust to fall back [...]

Great, I clone the repo and it can build successfully. It works, thanks!

closed 2023-01-31 by hzeller

RadicalRoman, 2023-04-27:
I run into this issue when trying to add verible as a submodule. I'm using the most recent commit: d600c4a77c001885ac342a3cf2cc66e55ef87273

PR #1659 (2023-01-31): Provide a fallback git date and version.
If git is not available and the GIT_DATE and GIT_VERSION environment variables are not set, fall back to a
generic string.

Fixes #1658
commit 88b1067f (2023-01-31): Provide a fallback git date and version.

If git is not available and the GIT_DATE and GIT_VERSION
environment variables are not set, fall back to a
generic string.

Fixes #1658