chipsalliance/verible #1666: Missing color scheme for verilog and fails while working with Verilog-HDL plugin

alexanderbrav, 2023-02-01:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ?
    vscode 
 - IDE version: 1-65-1
 - What other SystemVerilog plugins are active alongside ?
 : Verilog-HDL/SystemVerilog/Bluespec SystemVerilog
v1.7.0
-----

**What activity failed**
_(what functionality did not work as expected (e.g. Linting, Formatting, AUTO-expansion, go-to-definition, outline, hover,...)_
1. Without the verilog-HDL plugin the code coloring is missing.
2. While enabling the  verilog-HDL plugging the highlighting of the verilog style stops working.
**Expectation**
To support theme plugins or to provide color verilog theme.

**What actually happened**

**Test case**

```
module buff #(WIDTH=1) ();
endmodule
```

If needed, include SystemVerilog sample file(s) that show the code you want to do an operation on
```systemverilog
// Sample SystemVerilog file in case
```

**Logfiles**

If you invoke your editor with the environment variable `VERIBLE_LOGTHRESHOLD` set to 0, then useful log information will
be generated that you can include here

```bash
export VERIBLE_LOGTHRESHOLD=0
code foo.sv
```

***How to see the log output***
 - _vscode_: go to `View -> Output`, then choose `Verible Language Server` in the dropdown
 - _emacs_: there is a buffer `*verible-ls::stderr*` that contains the log output
 - _kate_: output is printed on the terminal

```txt
(paste log output here)
```

hzeller, 2023-02-01:
@tgorochowik to triage.

glatosinski, 2023-02-02:
passing the task to @jbylicki

closed 2023-02-02 by hzeller

hzeller, 2023-02-02:
This should start working with [v0.0.2821](https://github.com/chipsalliance/verible/releases/tag/v0.0-2821-gb2180bfa). It requires to install the new [verible.vsix](https://github.com/chipsalliance/verible/releases/download/v0.0-2821-gb2180bfa/verible.vsix):

So, to install/update the extension in VSCode:
```
wget -Overible.vsix https://github.com/chipsalliance/verible/releases/download/v0.0-2821-gb2180bfa/verible.vsix
code --install-extension verible.vsix
```

PR #1677 (2023-02-02): Changed VSCode plugin language capabilities
This PR changes how SystemVerilog capability is being communicated to VSCode, from `system-verilog` to just `systemverilog` - it is the default way, otherwise it produces conflicts with other packages and does not load the plugin alongside them.

Fixes #1666