chipsalliance/verible #1667: Format document duplicates last line.

alexanderbrav, 2023-02-01:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ? vscode
 - IDE version: 1-65-1
 - What other SystemVerilog plugins are active alongside ? None
-----

**What activity failed**
_(what functionality did not work as expected (e.g. Linting, Formatting, AUTO-expansion, go-to-definition, outline, hover,...)_
format document 
**Expectation**

**What actually happened**
while there is no empty line in the end of the verilog code the `format document` command duplicates last existing line.

**Test case**

Source code:
```
module InstModule (o,i);
      output [31:0] o;
      input i;
      wire [31:0] o = {32{i}};
  endmodule

  module ExampInst (o,i);

    output o;
    input i;

    InstModule instName
      (/*AUTOINST*/);

  endmodule
```

Result:
```
module InstModule (
    o,
    i
);
  output [31:0] o;
  input i;
  wire [31:0] o = {32{i}};
endmodule

module ExampInst (
    o,
    i
);

  output o;
  input i;

  InstModule instName (  /*AUTOINST*/);

endmodule
  endmodule
```

If needed, include SystemVerilog sample file(s) that show the code you want to do an operation on
```systemverilog
// Sample SystemVerilog file in case
```

**Logfiles**

If you invoke your editor with the environment variable `VERIBLE_LOGTHRESHOLD` set to 0, then useful log information will
be generated that you can include here

```bash
export VERIBLE_LOGTHRESHOLD=0
code foo.sv
```

***How to see the log output***
 - _vscode_: go to `View -> Output`, then choose `Verible Language Server` in the dropdown
 - _emacs_: there is a buffer `*verible-ls::stderr*` that contains the log output
 - _kate_: output is printed on the terminal

```txt
(paste log output here)
```

hzeller, 2023-02-01:
Assigned to @tgorochowik to triage.

hzeller, 2023-02-01:
Testing with `verible-verilog-format`, that does not happen. So must be something that is problematic when sending the change via json.

closed 2023-02-02 by hzeller

hzeller, 2023-02-02:
This fix is in [v0.0-2811](https://github.com/chipsalliance/verible/releases/tag/v0.0-2811-g6ce76fa8)

PR #1674 (2023-02-02): formatting: reply to language server client full original text range.
In case the text did not end with a newline, the end-position of text to replace in the formatting response did not include the remaining characters of the last line.

Fixes: #1667
commit a1f91435 (2023-02-02): formatting: reply to language server client full original text range.

In case the text did not end with a newline, the end-position of
text to replace in the formatting response did not include the
remaining characters of the last line.

Fixes: #1667