chipsalliance/verible #1668: Go definition doesn work

alexanderbrav, 2023-02-01:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ? vscode
 - IDE version: 1-65-1
 - What other SystemVerilog plugins are active alongside ? None
-----

**What activity failed**
_(what functionality did not work as expected (e.g. Linting, Formatting, AUTO-expansion, go-to-definition, outline, hover,...)_
`Go to Definition`
**Expectation**

**What actually happened**
The `Go To Definition` function doesn't work not for module instances not for  IO ports.
**Test case**

If needed, include SystemVerilog sample file(s) that show the code you want to do an operation on
```systemverilog
module InstModule (
    o,
    i
);
  output [31:0] o;
  input i;
  wire [31:0] o = {32{i}};
endmodule

module ExampInst (
    o,
    i
);

  output o;
  input i;

  InstModule instName (  /*AUTOINST*/);

endmodule
```

**Logfiles**

If you invoke your editor with the environment variable `VERIBLE_LOGTHRESHOLD` set to 0, then useful log information will
be generated that you can include here

```bash
export VERIBLE_LOGTHRESHOLD=0
code foo.sv
```

***How to see the log output***
 - _vscode_: go to `View -> Output`, then choose `Verible Language Server` in the dropdown
 - _emacs_: there is a buffer `*verible-ls::stderr*` that contains the log output
 - _kate_: output is printed on the terminal

```txt
(paste log output here)
```

hzeller, 2023-02-01:
CC @tgorochowik

hzeller, 2023-02-01:
Alex, did you have a `verible.filelist` ? This could be #1647 (which is already fixed after the release version).
At least for the module.

The IO ports might be a separate issue. Will hand over to Grzegorz to determine.

glatosinski, 2023-02-02:
I am taking a look at it

alexanderbrav, 2023-02-02:
I didn't use `verible.filelist`.

glatosinski, 2023-02-02:
The issue is being resolved in PR #1678 - I will provide the current status there

fangism, 2023-02-04:
Acknowledging that my initial implementation (~2 years ago) did not support this style of ports.

PR #1678 (2023-07-04): [LSP] textDocument/definition: Going to ports and module definitions
Fixes #1673
Fixes #1668
Fixes #1783

This PR:

- [X] Adds fix for finding definition of the module when cursor points at symbol definition
- [X] Adds fix for finding definition of ports for module - initial tests added, the solution seems to work.
- [X] Fixes the smoke tests which fail on "redefined" symbols, when symbol definition is fragmented.
- [X] Adds initial multiline goto-definition support to the LanguageServer
- [x] Implements tests for multiline goto-definition
- [x] Implements type checking for split definitions o [... cut]
closed 2023-07-04 by glatosinski