chipsalliance/verible #1669: Build error on Windows

LoveMyPillow, 2023-02-01:
The issue is an extension from the existed issue #958 . However, it seems that the build guidance on windows is still not released, so I open a new issue here.

I try to compile verible on windows10, developer mode as #796 mentioned, and use wsl2 to run bash on windows. When I run the command:
`bazel build -c opt --enable_runfiles //...`
or
`bazel test --test_output=errors //...`
 
the console print as below:
//================================================================
ERROR: D:/s-verible/verible/common/util/BUILD:86:8: Executing genrule //common/util:version_header [for tool] failed: (Exit 127): bash.exe failed: error executing command (from target //common/util:version_header) C:\Windows\system32\bash.exe -c ... (remaining 1 argument skipped)
/usr/bin/env: ‘bash\r’: No such file or directory
//================================================================
It seems like something about CRLF and CR(I guess, as mention on #796 ), and I don't think I am on the right way to compile this project on windows...
Thanks!

hzeller, 2023-02-01:
I've simplified a few things, can you try the latest version in git and see if the compile proceeds further ?

LoveMyPillow, 2023-02-02:
> I've simplified a few things, can you try the latest version in git and see if the compile proceeds further ?

yes, the compile proceeds further, but it meets new errors. And as to the bash shell and LLVM toolchain, I have installed them before this issue opened.

The error log is attached.

//=============================================
D:\S-Verible\verible>bazel build -c opt --enable_runfiles //...
INFO: Analyzed 1894 targets (126 packages loaded, 6330 targets configured).
INFO: Found 1894 targets...
INFO: From Executing genrule //verilog/parser:verilog_lex:
verilog/parser/verilog.lex:305: warning, rule cannot be matched
verilog/parser/verilog.lex:971: warning, rule cannot be matched
ERROR: D:/s-verible/verible/verilog/parser/BUILD:110:24: Executing genrule //verilog/parser:verilog_y_moved failed: (Exit 1): bash.exe failed: error executing command (from target //verilog/parser:verilog_y_moved) C:\Windows\system32\bash.exe -c ... (remaining 1 argument skipped)
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe
write failed 32: Broken pipe

LoveMyPillow, 2023-02-02:
Do you mean drop the first line and the content of this line "#! /bin/sh"? Yes, the warning message disappeared but error message remained.

hzeller, 2023-02-02:
yes, the first line that contains the `/bin/sh`

What is the output if you run

```
bazel build verilog/parser:verilog_y_moved --sandbox_debug
```

LoveMyPillow, 2023-02-02:
`D:\S-Verible\verible>bazel build verilog/parser:verilog_y_moved --sandbox_debug
INFO: Build options --compilation_mode and --enable_runfiles have changed, discarding analysis cache.
INFO: Analyzed target //verilog/parser:verilog_y_moved (0 packages loaded, 161 targets configured).
INFO: Found 1 target...
ERROR: D:/s-verible/verible/verilog/parser/BUILD:110:24: Executing genrule //verilog/parser:verilog_y_moved failed: (Exit 1): bash.exe failed: error executing command (from target //verilog/parser:verilog_y_moved) C:\Windows\system32\bash.exe -c ... (remaining 1 argument skipped)
write failed 32: Broken pipe
write failed 32: Broken pipe
//too many duplicated meesage and I hide them :)
write failed 32: Broken pipe
Target //verilog/parser:verilog_y_moved failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 8.383s, Critical Path: 7.68s
INFO: 5 processes: 4 internal, 1 local.
FAILED: Build did NOT complete successfully`

And I noticed that the program suggest me to use --verbose_failures option to see the error message, and I attached the result here. Hope this can do some help.
`D:\S-Verible\verible>bazel build verilog/parser:verilog_y_moved --sandbox_debug --verbose_failures
INFO: Analyzed target //verilog/parser:verilog_y_moved (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
ERROR: D:/s-verible/verible/verilog/parser/BUILD:110:24: Executing genrule //verilog/parser:verilog_y_moved failed: (Exit 1): bash.exe failed: error executing command (from target //verilog/parser:verilog_y_moved)
  cd /d C:/users/10268/_bazel_10268/mxjvn6ym/execroot/com_google_verible
  SET PATH=C:\Windows\system32;C:\Windows;C:\Windows\System32;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;D:\S-Git\cmd;C:\Program Files\Wolfram Research\WolframScript\;D:\S-Bazel;D:\S-Python\Python310\Scripts\;D:\S-Python\Python310\;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps;;D:\intelFPGA\20.1\ModelSim\modelsim_ase\win32aloem;D:\S-Microsoft VS Code\bin;D:\S-Wind.NET.Client\WindNET\bin;D:\S-Wind.NET.Client\WindNET\x64
    SET RUNFILES_MANIFEST_ONLY=1
  C:\Windows\system32\bash.exe -c source external/bazel_tools/tools/genrule/genrule-setup.sh; bazel-out/x64_windows-opt-exec-2B5CBBC6/bin/common/parser/move_yacc_stack_symbols.exe < bazel-out/x64_windows-fastbuild/bin/verilog/parser/verilog.tab.cc > bazel-out/x64_windows-fastbuild/bin/verilog/parser/verilog-moved.tab.cc
# Configuration: b26347f9d87ef03821af840fb9b4d879c23c93cd30e48cee845dbeefe1752b5a
# Execution platform: @local_config_platform//:host
write failed 32: Broken pipe
write failed 32: Broken pipe
//too many duplicated meesage and I hide them :)
write failed 32: Broken pipe
Target //verilog/parser:verilog_y_moved failed to build
INFO: Elapsed time: 1.013s, Critical Path: 0.59s
INFO: 2 processes: 2 internal.
FAILED: Build did NOT complete successfully`

hzeller, 2023-02-02:
mmh, I am wondering why the sh_binary(), which is just a shellscript, becomes `move_yacc_stack_symbols.exe`; maybe that is what bazel generates on Windows.

Maybe that works not well on that platform. I'll  see if I can formulate that inline as sed script that worked with the version_header before. Will be back at the computer in an hour.

corco, 2023-02-02:
You may want to take a look at [Github's Windows configuration](https://github.com/actions/runner-images/blob/main/images/win/Windows2022-Readme.md). It has git bash, msys2 bash and WSL bash installed.

I don't remember how the machine I used was configured (I don't use Windows myself...), but I'm pretty sure the bash shell it was using was git bash or MSYS2 bash and not the WSL one... Bazel does a lot of stuff with shell script on Windows, such as generate .exe from them as @hzeller pointed out; it may simply not work in a WSL context.

Bazel itself seems to have (issue with WSL)[https://github.com/bazelbuild/bazel/issues/10941]

LoveMyPillow, 2023-02-02:
> You may want to take a look at [Github's Windows configuration](https://github.com/actions/runner-images/blob/main/images/win/Windows2022-Readme.md). It has git bash, msys2 bash and WSL bash installed.
> 
> I don't remember how the machine I used was configured (I don't use Windows myself...), but I'm pretty sure the bash shell it was using was git bash or MSYS2 bash and not the WSL one... Bazel does a lot of stuff with shell script on Windows, such as generate .exe from them as @hzeller pointed out; it may simply not work in a WSL context.
> 
> Bazel itself seems to have (issue with WSL)[https://github.com/[bazelbuild/bazel/issues/10941](https://github.com/bazelbuild/bazel/issues/10941)]

Oh, it seems to point out the key, I will try another bash today.

LoveMyPillow, 2023-02-02:
When I use git bash by set the windows environment variable `BAZEL_SH` to git bash.exe, the project was built successfully and all 568 tests invoked by `build test -c opt //...` passed.

During compilation, some warning messages were printed:
//============================Part of warning messages==============================
INFO: From Compiling adler32.c [for tool]:
clang-cl: warning: argument unused during compilation: '/std:c++17' [-Wunused-command-line-argument]
INFO: From Compiling deflate.c [for tool]:
clang-cl: warning: argument unused during compilation: '/std:c++17' [-Wunused-command-line-argument]
INFO: From Compiling gzwrite.c [for tool]:
clang-cl: warning: argument unused during compilation: '/std:c++17' [-Wunused-command-line-argument]
INFO: From Compiling inffast.c [for tool]:
clang-cl: warning: argument unused during compilation: '/std:c++17' [-Wunused-command-line-argument]
INFO: From Linking external/com_google_protobuf/protoc.exe [for tool]:
lld-link: warning: ignoring unknown argument '-lpthread'
lld-link: warning: ignoring unknown argument '-lm'
lld-link: warning: ignoring unknown argument '-lpthread'
lld-link: warning: ignoring unknown argument '-lm'
lld-link: warning: ignoring unknown argument '-lpthread'
lld-link: warning: ignoring unknown argument '-lm'
lld-link: warning: ignoring unknown argument '-lpthread'
lld-link: warning: ignoring unknown argument '-lm'
//===========================================================================

It seems that these warnings don't affect the test result. Thank you all.

hzeller, 2023-02-02:
In #1675 , I skipped using shell scripts for the file modifications but directly do it in the starlark script, which hopefully simplifies it enough to compile hopefully without the `BAZEL_SH` change you had to do (though shell tests that heavily rely on a working shell might still not work).

Would you like to write a small paragraph w.r.t. to compilation on Windows (what to install, what environment variables to set etc.) that we can put in the [developers section of the README](https://github.com/chipsalliance/verible/blob/master/README.md#developers-welcome) @LoveMyPillow  ? So that the next person compiling it on Windows can get it working quickly.

LoveMyPillow, 2023-02-03:
The following paragraph is what I try to do for building Verible, I don't know if JAVA jdk, python3, gcc and others are needed because I saw them on shell.nix but the .bazelrc specified the compiler to clang-cl, so they were omitted by me.
And I have to admit that English is my native language, maybe you will find some grammer errors, :)
//=====================================================================
To build Verible on Windows, you need:
- git [version: latest] and the `BAZEL_SH` environment variable
	 `set BAZEL_SH = C:\Program Files\Git\bin\bash.exe`
- Visual C++ build tool and Windows 10 SDK. (Visual Studio Installer)[version: 2019 community]
	 you need Visual c++ bulid tools (2017 or higher version) and Windows 10 SDK to make clang works on bazel.
	 if you want to specify the VC installation, use the `BAZEL_VC` and `BAZEL_VC_FULL_VERSION` environment 
		 `set BAZEL_VC = C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC`
		 `set BAZEL_VC_FULL_VERSION = 14.29.30133`
	 if you want to specify the SDK version, use the `BAZEL_WINSDK_FULL_VERSION` environment variable
		 `set BAZEL_WINSDK_FULL_VERSION = 10.0.19041.0`
- LLVM[version 12.0.0] and `BAZEL_LLVM` environment variable
	 `set BAZEL_LLVM = C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\Llvm\x64`
	 *Be care: you should choose the x64 version*
- Bazel [version 7.0.0] and added it into `PATH` envrionment variable
According to pull request #796 , you should enable developer mode on Windows and use`bazel build -c opt --enable_runfiles //...` to build the project.

durongze, 2023-07-29:
WARNING: Ignoring JAVA_HOME, because it must point to a JDK, not a JRE.
ERROR: C:/users/administrator/_bazel_administrator/rs3thaef/external/local_config_sh/BUILD:6:14: syntax error at 'C': expected ,
INFO: Repository rules_ruby instantiated at:
  D:/code/verilog/verible/WORKSPACE:152:14: in <toplevel>
  C:/users/administrator/_bazel_administrator/rs3thaef/external/com_google_protobuf/protobuf_deps.bzl:109:24: in protobuf_deps
  C:/users/administrator/_bazel_administrator/rs3thaef/external/com_google_protobuf/protobuf_deps.bzl:20:17: in _github_archive
Repository rule http_archive defined at:
  C:/users/administrator/_bazel_administrator/rs3thaef/external/bazel_tools/tools/build_defs/repo/http.bzl:372:31: in <toplevel>
INFO: repository @rules_ruby' used the following cache hits instead of downloading the corresponding file.
 * Hash 'c88dd69eb50fcfd7fbc5d7db79adc6631ef0e1d80b3c94efe33ac5ee3ccc37f7' for https://github.com/protocolbuffers/rules_ruby/archive/5cf6ff74161d7f985b9bf86bb3c5fb16cef6337b.zip
If the definition of 'repository @rules_ruby' was updated, verify that the hashes were also updated.
ERROR: D:/code/verilog/verible/bazel/BUILD:14:11: While resolving toolchains for target //bazel:flex: invalid registered toolchain '@local_config_sh//:local_sh_toolchain': no such target '@local_config_sh//:local_sh_toolchain': target 'local_sh_toolchain' not declared in package '' defined by C:/users/administrator/_bazel_administrator/rs3thaef/external/local_config_sh/BUILD (Tip: use `query "@local_config_sh//:*"` to see all the targets in that package)
ERROR: Analysis of target '//bazel:flex' failed; build aborted
INFO: Elapsed time: 1.004s
INFO: 0 processes.
ERROR: Build did NOT complete successfully
Loading: 4 packages loaded
    currently loading: @com_google_protobuf// ... (6 packages)
    Fetching repository @win_flex_bison; starting
    Fetching repository @rules_pkg; starting
    Fetching C:/users/administrator/_bazel_administrator/rs3thaef/external/win_flex_bison; Extracting win_flex_bison-2.5.25.zip
    Fetching C:/users/administrator/_bazel_administrator/rs3thaef/external/rules_pkg; Extracting rules_pkg-0.7.0.tar.gz
    Fetching C:/users/administrator/_bazel_administrator/rs3thaef/external/rules_ruby; Extracting 5cf6ff74161d7f985b9bf86bb3c5fb16cef6337b.zip
    Fetching repository @python_anytree; starting
    Fetching C:/users/administrator/_bazel_administrator/rs3thaef/external/python_anytree; Extracting 2.8.0.tar.gz

hzeller, 2023-07-29:
The Verible binaries are regularly compiled in the CI using the following set-up:

https://github.com/chipsalliance/verible/blob/master/.github/workflows/verible-ci.yml#L481-L512

Here is a typical CI run https://github.com/chipsalliance/verible/actions/runs/5673921578/job/15376447668

If it is not working for you, maybe you can compare and see where your set-up differs ?

corco, 2023-08-09:
As far as I recall, WSL2 bash is not supported by bazel. You need a Windows-flavored bash shell like msys2.

In development, I used powershell to start bazel, and it detected/used the bash shell provided by Git Bash for bash scripts.

liuyd-own, 2025-06-06:
PS D:\Liuyd\Programs\Verilog\verible-master> bazel build --sandbox_debug --verbose_failures --subcommands --config=windows -c opt //...
Starting local Bazel server and connecting to it...
ERROR: <builtin>: BazelWorkspaceStatusAction stable-status.txt failed: Failed to determine workspace status: Process exited with status 1
INFO: Elapsed time: 9.997s, Critical Path: 0.26s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully

liuyd-own, 2025-06-06:
PS D:\Liuyd\Programs\Verilog\verible-master> choco list
Chocolatey v2.4.3
bazel 7.6.1
chocolatey 2.4.3
chocolatey-compatibility.extension 1.0.0
chocolatey-core.extension 1.4.0
chocolatey-windowsupdate.extension 1.0.5
git 2.49.0
git.install 2.49.0
jdk8 8.0.211
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
llvm 18.1.8
python311 3.11.9
vcredist140 14.44.35208
vcredist2015 14.0.24215.20170201
winflexbison3 2.5.24.20210105
18 packages installed.

liuyd-own, 2025-06-09:
Could you please tell me the cause of the following error?Thanks

PS D:\Liuyd\Programs\Verilog\verible-master> bazel build -c opt //...
WARNING: Download from https://github.com/bazelbuild/bazel-skylib/releases/download/1.7.1/bazel-skylib-1.7.1.tar.gz failed: class javax.net.ssl.SSLHandshakeException PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
INFO: Repository bazel_skylib~ instantiated at:
  <builtin>: in <toplevel>
Repository rule http_archive defined at:
  C:/users/liuyd/_bazel_liuyd/pskq4vks/external/bazel_tools/tools/build_defs/repo/http.bzl:387:31: in <toplevel>
ERROR: C:/users/liuyd/_bazel_liuyd/pskq4vks/external/bazel_tools/tools/build_defs/repo/http.bzl:136:45: An error occurred during the fetch of repository 'bazel_skylib~':
   Traceback (most recent call last):
        File "C:/users/liuyd/_bazel_liuyd/pskq4vks/external/bazel_tools/tools/build_defs/repo/http.bzl", line 136, column 45, in _http_archive_impl
                download_info = ctx.download_and_extract(
Error in download_and_extract: java.io.IOException: Error downloading [https://github.com/bazelbuild/bazel-skylib/releases/download/1.7.1/bazel-skylib-1.7.1.tar.gz] to C:/users/liuyd/_bazel_liuyd/pskq4vks/external/bazel_skylib~/temp6204947854199552293/bazel-skylib-1.7.1.tar.gz: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
WARNING: Target pattern parsing failed.
ERROR: Skipping '//...': error loading package under directory '': no such package '@@bazel_skylib~//rules': java.io.IOException: Error downloading [https://github.com/bazelbuild/bazel-skylib/releases/download/1.7.1/bazel-skylib-1.7.1.tar.gz] to C:/users/liuyd/_bazel_liuyd/pskq4vks/external/bazel_skylib~/temp6204947854199552293/bazel-skylib-1.7.1.tar.gz: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
ERROR: error loading package under directory '': no such package '@@bazel_skylib~//rules': java.io.IOException: Error downloading [https://github.com/bazelbuild/bazel-skylib/releases/download/1.7.1/bazel-skylib-1.7.1.tar.gz] to C:/users/liuyd/_bazel_liuyd/pskq4vks/external/bazel_skylib~/temp6204947854199552293/bazel-skylib-1.7.1.tar.gz: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
INFO: Elapsed time: 12.399s
INFO: 0 processes.
ERROR: Build did NOT complete successfully
FAILED:
    Fetching repository @@protobuf~; starting 11s
    Fetching repository @@rules_license~; starting 11s