chipsalliance/verible #1682: SymbolTable: update incrementally ?

hzeller, 2023-02-03:
Given a large file list with only a few files changed and the need to update the SymbolTable: can we do it without having to re-index everything, but just update symbols incrementally ?

This could help for larger projects, in particular when we have to wait in the language server when calling 'go to definition' / 'find references'.

There are multiple parts to this question

  * How big of a problem is this ? Given a huge project, what percentage of time is spent on parsing vs. SymbolTable update.
  * Given a large project, what is the absolute time in each of these spent ? (Ideally, we would like to minimize 99% time for user queries to < 500ms).
  * How much complexity would this add - is it worthwhile vs. maintenance overhead ?

hzeller, 2023-02-03:
CC @fangism in case he has some ideas w.r.t. complexity as he implemented the symbol table.

hzeller, 2023-02-03:
Asked David, and his first assessment is that it might be a larger task, probably requires some careful measurement first how much this is a problem.

fangism, 2023-02-04:
One difficulty with this is that the symbol table is a monolithic accumulated structure, whose result is the accumulation of changes from processing multiple files.  There could be interactions between files as a result of one file being updated, one moved definition could alter the implicit dependency graph (of files).  For this idea to work, you would need some well-defined state of the symbol-table between processing each translation unit, and either transitions between states have to be reversible, or you'd keep around copies of old state for reversibility.  This problem maps naturally to incremental builds systems, whose primary purpose is to know precisely what subgraph needs to be redone for any update.