chipsalliance/verible #1690: LSP fails go-to-definition in Windows environment

viktortomov, 2023-02-07:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ? 
 - IDE version: neovim v0.8.3
 - What other SystemVerilog plugins are active alongside ?
 - mason with verible lsp installed
-----

**What activity failed**
_(what functionality did not work as expected (e.g. Linting, Formatting, AUTO-expansion, go-to-definition, outline, hover,...)_
go-to-definition

**Expectation**
go to the specific line of the definition

**What actually happened**
Seems that the nvim lsp is giving an error due to incorrect url from the verible language server. See the last debug message in the log file. The url starts with ```file://C:\\C:\\``` where it should be  ```file:///C:/workspace/.....```

**Test case**
The cursor is under the cntr assignment and the go to function is executed.

If needed, include SystemVerilog sample file(s) that show the code you want to do an operation on
```systemverilog
// Sample SystemVerilog file in case
// Simple counter for verification purposes.

module smpl_cntr
(
    input logic clk,
    output logic cntr
);

always_ff @(posedge clk) begin
    cntr <= cntr + 1'b1;
end

endmodule
```

**Logfiles**

```txt
[START][2023-02-07 18:26:30] LSP logging initiated
[ERROR][2023-02-07 18:26:30] .../vim/lsp/rpc.lua:733	"rpc"	"C:\\Users\\victor.tomov\\AppData\\Local\\nvim-data\\mason\\bin\\verible-verilog-ls.CMD"	"stderr"	"Verible Verilog Language Server built at v0.0-2832-gffdd6671\r\n"
[ERROR][2023-02-07 18:26:30] .../vim/lsp/rpc.lua:733	"rpc"	"C:\\Users\\victor.tomov\\AppData\\Local\\nvim-data\\mason\\bin\\verible-verilog-ls.CMD"	"stderr"	"E0207 16:26:30.297135   23252 json-rpc-dispatcher.cc:58] Unhandled notification 'initialized'\r\n"
[DEBUG][2023-02-07 18:26:40] .../lua/vim/lsp.lua:1381	"LSP[verible]"	"client.request"	1	"textDocument/definition"	{  position = {    character = 6,    line = 9  },  textDocument = {    uri = "file:///C:/workspace/git/tool_validation/rtl/smpl_cntr/smpl_cntr.sv"  }}	<function 1>	1
[DEBUG][2023-02-07 18:26:40] .../vim/lsp/rpc.lua:285	"rpc.send"	{  id = 2,  jsonrpc = "2.0",  method = "textDocument/definition",  params = {    position = {      character = 6,      line = 9    },    textDocument = {      uri = "file:///C:/workspace/git/tool_validation/rtl/smpl_cntr/smpl_cntr.sv"    }  }}
[DEBUG][2023-02-07 18:26:40] .../vim/lsp/rpc.lua:388	"rpc.receive"	{  id = 2,  jsonrpc = "2.0",  result = { {      range = {        ["end"] = {          character = 21,          line = 5        },        start = {          character = 17,          line = 5        }      },      uri = "file://C:\\C:\\workspace\\git\\tool_validation\\rtl\\smpl_cntr\\smpl_cntr.sv"    } }}

```

zhuzhzh, 2023-02-08:
I hit the same issue with 2847.

zhuzhzh, 2023-02-09:
with 2858, it exists as well on centos7.9.

```txt
Verible Verilog Language Server built at v0.0-2858-g0f9f5f21
E0209 10:54:07.027395   60160 json-rpc-dispatcher.cc:58] Unhandled notification 'initialized'
E0209 10:54:07.034828   60160 json-rpc-dispatcher.cc:58] Unhandled notification '$/setTrace'
E0209 10:54:08.032300   60160 json-rpc-dispatcher.cc:58] Unhandled notification '$/setTrace'
E0209 10:54:08.555797   60160 json-rpc-dispatcher.cc:58] Unhandled notification '$/setTrace'
E0209 10:54:10.027790   60160 json-rpc-dispatcher.cc:58] Unhandled notification '$/cancelRequest'
E0209 10:54:11.346671   60160 json-rpc-dispatcher.cc:58] Unhandled notification '$/setTrace'
E0209 10:54:11.957739   60160 json-rpc-dispatcher.cc:58] Unhandled notification '$/setTrace'
```

hzeller, 2023-02-09:
Please file separate issues for different observations.

 * The original issue seems to be an issue file name concatenation as Windows has drive letters.
 * The 'unhandled notification' are benign. It just says that it ignores non-relevant notification (notifications are essentially FYI messages). The question is: does the resolve of the symbol work ?

Do you have a file-list ? The language server wants a `verible.filelist`, otherwise it will only look at the current file.

If you want to increase the log-level, set the environment variable `VERIBLE_LOGTHRESHOLD=0`

hzeller, 2023-02-09:
Grzegorz, can you have a look at assembly of windows file-paths ?

zhuzhzh, 2023-02-09:
@hzeller Go-to-defination doesn't work at begining. I thought it's related to those error messages.  But now it works . I used verible.filelist. I use `VERIBLE_LOGTHRESHOLD=0` to confirm it works.

hzeller, 2023-02-09:
Nice @zhuzhzh  Good that the `VERIBLE_LOGTHRESHOLD` helped in confirming what is going on under the hood.