chipsalliance/verible #1713: AUTOPorts sometimes 'out of order', making the tests fail

hzeller, 2023-02-15:
When importing verible into our environment I get some tests to be flaky since #1691. Symptom is that emitted ports are out of order from the expected order, in some of the tests (in `autoexpand_test`), and not always the same tests.

So running the `verilog/tools/ls:autoexpand_test` will  have randomly a few of the tests fail (see below for an example).

I _suspect_ that is because we're using a different hash function for each run to make sure there is nothing depending on the iteration sequence of hash functions. Could it be something like that ?
Or maybe string views are compared from different sources, so that they are sensitive to allocation order ?

Unfortunately, I can't reproduce this in the github CI.
But since you're deeply in that code, maybe you have an idea where to look @kbieganski ?

```console
 RUN      ] Autoexpand.AUTO_ExpandPorts
verilog/tools/ls/autoexpand_test.cc:79: Failure
Expected equality of these values: (...)
With diff:
@@ -8,9 +8,9 @@
 module foo(/*AUTOARG*/
   // Inputs
-  i2, i1,
+  i1, i2,
   // Inouts
   io,
   // Outputs
-  o2, o1
+  o1, o2
   );
   /*AUTOINPUT*/
[.. similar with other tests..]
[ RUN      ] Autoexpand.CodeActionExpandAll
verilog/tools/ls/autoexpand_test.cc:79: Failure
Expected equality of these values: (...)
With diff:
@@ -2,7 +2,7 @@
 module foo(/*AUTOARG*/
   // Inputs
-  rst, clk,
+  clk, rst,
   // Outputs
-  o2, o1
+  o1, o2
   );
   /*AUTOINPUT*/
```

kbieganski, 2023-02-15:
Is the discrepancy always in `/*AUTOARG*/`?

hzeller, 2023-02-15:
Ran it a couple of times, and these seem to be the tests that regularly fail

```console
Autoexpand.AUTO_ExpandPorts
Autoexpand.AUTO_ExpandPorts_OutOfOrderModules
Autoexpand.AUTO_ExpandPortsWithAUTOVars
Autoexpand.CodeActionExpandAll
Autoexpand.CodeActionExpandRange
```

kbieganski, 2023-02-15:
The best guess I have is this: autogenerated ports from one 'category' (e.g. `INPUT`) are all assigned the same byte offset, which is at the start of the `AUTO` comment. Later, ports are sorted by the offset, but this is done by `std::sort`, which is **NOT** a stable sort. Thus resulting in different order depending on what pivot is picked or whatever.

Could you try this patch:
```diff
diff --git a/verilog/tools/ls/autoexpand.cc b/verilog/tools/ls/autoexpand.cc
index 679b5010..b53e5826 100644
--- a/verilog/tools/ls/autoexpand.cc
+++ b/verilog/tools/ls/autoexpand.cc
@@ -524,10 +524,10 @@ void AutoExpander::Module::MarkUndeclaredPortsAsAutogenerated(
 }
 
 void AutoExpander::Module::SortPortsByOffset() {
-  std::sort(ports_.begin(), ports_.end(),
-            [](const Port &left, const Port &right) {
-              return left.offset < right.offset;
-            });
+  std::stable_sort(ports_.begin(), ports_.end(),
+                   [](const Port &left, const Port &right) {
+                     return left.offset < right.offset;
+                   });
 }
 
 void AutoExpander::Module::RetrieveAutoTemplates() { 
```

hzeller, 2023-02-15:
Yes, that works. Good find!

PR #1714 (2023-02-15): Use stable sorting for AUTO expansion
Makes `SortPortsByOffset()` use stable sort. This is relevant for ports autogenerated via `AUTOINPUT`, `AUTOOUTPUT`, and `AUTOINOUT`. Ports in each of these groups get assigned the same offset, which is the start of the corresponding `AUTO` comment. Using unstable sort results in a random order of ports in an `AUTOARG` port list.

Fixes #1713.
commit f8083a5f (2023-02-15): Use stable sorting for AUTO expansion

Makes `SortPortsByOffset()` use stable sort. This is relevant for ports
autogenerated via `AUTOINPUT`, `AUTOOUTPUT`, and `AUTOINOUT`. Ports in
each of these groups get assigned the same offset, which is the start of
the corresponding `AUTO` comment. Using unstable sort results in a
random order of ports in an `AUTOARG` port list.

Fixes #1713.
closed 2023-02-15 by hzeller