chipsalliance/verible #1765: Formatter doesn't handle packed arrays of user defined types

shareefj, 2023-03-01:
I've just had a first go at using the formatter in VSCode and it seems to work nicely.  The issue I've found in the first few minutes of testing is that it doesn't seem to like packed arrays of user types.

The following example works cleanly with `module_net_variable_alignment=align`:

```systemverilog
module testcase (
  input logic clk,
  input logic rst_n,
  input logic [10:0] a,
  input my_special_t b
);

  logic [5:0] net_a;
  my_special_t net_b;
  logic [5:0] [5:0] net_c;

endmodule
```

```systemverilog
module testcase (
  input logic               clk,
  input logic               rst_n,
  input logic        [10:0] a,
  input my_special_t        b
);

  logic        [5:0]      net_a;
  my_special_t            net_b;
  logic        [5:0][5:0] net_c;

endmodule
```

However, if I add a packed array of a user defined type, things get a little confused.

```systemverilog
module testcase (
  input logic clk,
  input logic rst_n,
  input logic [10:0] a,
  input my_special_t b
);

  logic [5:0] net_a;
  my_special_t net_b;
  logic [5:0] [5:0] net_c;

  logic net_d;
  an_other_t [1:0] net_e;

endmodule
```

```systemverilog
module testcase (
  input logic               clk,
  input logic               rst_n,
  input logic        [10:0] a,
  input my_special_t        b
);

  logic          [5:0]          net_a;
  my_special_t                  net_b;
  logic          [5:0]   [5:0]  net_c;

  logic                         net_d;
  an_other_t   [      1:0     ] net_e;

endmodule

```

And in the larger real world example, I get lots of unaligned leading/trailing square brackets rather than the expected columns.

skazarynau1, 2023-04-04:
I have the same issue with user defined types

**Actual output**
```
module m1 ();
  typedef logic [31 : 0] uint32_t;
  logic      [0 : 5]     [15 : 0]  s1;
  uint32_t [        0 : 5        ] s2;
endmodule
```

**Expected/desired output**
```
module m1 ();
  typedef logic [31 : 0] uint32_t;
  logic    [0 : 5][15 : 0] s1;
  uint32_t [0 : 5]         s2;
endmodule
```

The following options are used
```
--inplace=true
--compact_indexing_and_selections=false
```

```
verible-verilog-format --version
v0.0-3124-gd7297f5b
Commit  2023-03-31
Built   2023-04-01T14:44:10Z
```

closed 2023-06-20 by hzeller

shareefj, 2023-06-21:
Thanks. Will try it again.

PR #1938 (2023-06-20): Adapt ClassProperty scanner dimension logic to declaration aligning
This PR adapts the `ClassPropertyColumnSchemaScanner` dimension aligning logic into `DataDeclarationColumnSchemaScanner` as it produced malformed outputs when formatting a custom type.

Fixes #1765