chipsalliance/verible #1770: How to format task and functions parameters the same as ansi module ports?

prokie, 2023-03-06:
**Test case**

```systemverilog
module ansi_module (
  input var logic a  /* input a */,
  input var logic b  /* input b */,
  input var logic c  /* input c */,
  inout tri logic d  /* input d */
);
  function test(input logic test_a  /* input test_a */, input logic test_b  /* input test_a */,
                input logic test_c /* input test_a */
                   );
endfunction
endmodule
```
I use this configuration, but is possible to configure `verible-verilog-format` to get the same format for the params in functions and tasks as I get for ports in the module declaration?
`verible-verilog-format --wrap_spaces 2 --port_declarations_alignment flush-left`

**Actual output**

```systemverilog
module ansi_module (
  input var logic a  /* input a */,
  input var logic b  /* input b */,
  input var logic c  /* input c */,
  inout tri logic d  /* input d */
);
  function test(input logic test_a  /* input test_a */, input logic test_b  /* input test_b */,
                input logic test_c /* input test_c */
                   );
  endfunction
endmodule
```

**Expected or suggested output**

```systemverilog
module ansi_module (
  input var logic a  /* input a */,
  input var logic b  /* input b */,
  input var logic c  /* input c */,
  inout tri logic d  /* input d */
);
  function test(
    input logic test_a  /* input test_a */,
    input logic test_b  /* input test_b */,
    input logic test_c  /* input test_c */
  );
  endfunction
endmodule
```

hzeller, 2023-03-06:
I think there is no configuration option yet, but might be worthwile.
@tgorochowik to triage,
@mglb might be able to point to the right place where to do.

kbrunham-intel, 2026-09-22:
@hzeller This looks like a duplicate of #1286. I would suggest closing this issue in favor of that one so we keep a single tracking ticket.