chipsalliance/verible #1775: Autoexpand crash when module name not found.

hzeller, 2023-03-07:
Playing around in some files (with possibly incomplete symbol table as the file list was not complete) I noticed crashes of the language server when it came around a situation in which `GetModuleName()` returned a nullptr. When given such symbol to a `Module()` object, it crashes as it dereferences that around line 145 in the `Module` constructor.

Testing for that situation beforehand helped that case (quick hack patch below), but it then crashes later (still investigating).
So it would be good to have a test that creates a situation like this and test that things are robust afterwards.

```patch
--- a/verilog/tools/ls/autoexpand.cc
+++ b/verilog/tools/ls/autoexpand.cc
@@ -981,6 +981,10 @@ std::optional<AutoExpander::Expansion> AutoExpander::ExpandAutoinst(
     LOG(ERROR) << "AUTOINST: No definition found for module type: " << type_id;
     return {};
   }
+  if (!GetModuleName(*type_def)) {
+    LOG(ERROR) << "AUTOINST: No module name " << type_id;
+    return {};
+  }
   if (!modules_.contains(type_id)) {
     modules_.insert(std::make_pair(type_id, Module(*type_def)));
   }
```

kbieganski, 2023-03-07:
Not sure how to reproduce this situation in a test. If a module is undefined, the symbol table simply returns a null symbol pointer. The meaning of that is clear: the symbol was not found, and it's already checked against. For the module name being null, I'm guessing the symbol table returns a partially-initialized symbol while it's being built? I don't think we want that to happen at all. I'll look at the symbol table source, maybe something else is happening.

hzeller, 2023-03-09:
Alright, found the situation: it happens when a name of the module and the instance name is the same, like below `foo foo();`
Apparently that is a thing.

```SystemVerilog
module foo();
endmodule

module bar();
  foo foo(/*AUTOINST*/);
endmodule
```

kbieganski, 2023-03-09:
Thanks for the investigation, opened a PR that fixes this.

In your case the symbol table returns the instance itself, instead of the module. Which is surprising, I think it's being searched from the root? Maybe it's because it's DFS instead of BFS?

PR #1788 (2023-03-09): Handle incorrect instance types in `AUTOINST`
The `AUTOINST` expansion function assumes the symbol found by searching for the instance type name is a module declaration. However, if it's not a module declaration, Verible crashes due to theie assumption. This patch fixes the issue by checking the type of the node found by the symbol table.

Fixes #1775.
commit 8b5b0c26 (2023-03-09): Handle incorrect instance types in `AUTOINST`

The `AUTOINST` expansion function assumes the symbol found by searching for the
instance type name is a module declaration. However, if it's not a module
declaration, Verible crashes due to theie assumption. This patch fixes the issue
by checking the type of the node found by the symbol table.

Fixes #1775.
closed 2023-03-09 by hzeller