chipsalliance/verible #1783: Handling split definitions of ports in the Symbol Table

glatosinski, 2023-03-08:
**Description**

Currently, in the SymbolTable we have a MapTree of symbols (dependent on scoping), where the key in the map is a `string_view` with the symbol name and its location (based on pointers' distances).

In the case of ports, especially non-ANSI port definitions can be one-line, or split into multiple lines (as shown in the example below), where one line defines direction, other type etc. forming a full definition of the symbol.

The questions is: If (and how) SymbolTable should handle such multi-line definitions.

Things that should be discussed/considered are:

* Storing/representing multiple definitions
* Error-checking against providing conflicting information in various lines, e.g. multiple types provided for the symbol.

To reproduce the current behavior of the SymbolTable, let's create a `test-project` directory (let's store its path in `$TEST_PROJECT`) with a following file, named `add32.sv` (a full file can be found in [ivtest/contrib/add32.v](https://github.com/steveicarus/ivtest/blob/master/contrib/add32.v):

```systemverilog
module add32(sum, cOut, clock, a, b, cIn);

  input clock;
  input a, b, cIn;
  output sum, cOut;

  reg [31:0] a, b;
  reg cIn;
  wire [31:0] sum;
  wire cOut;

  always @(posedge clock)
    //{cOut, sum} = a + b + cIn;
    assign sum = a + b + cIn;

endmodule
```

and a `verible.filelist` file:
```
add32.sv
```

With the directory with those two files, run `verible-verilog-project`:

```bash
verible-verilog-project symbol-table-defs --file_list_root ${TEST_PROJECT} --file_list_path ${TEST_PROJECT}/verible.filelist
```

**Actual behavior:**

Symbol Table is printed as follows:

```
Symbol Table:
{ (
    metatype: <root>
)
  add32: { (
      metatype: module
      file: /test-project/add32.sv
  )
    a: { (
        metatype: data/net/var/instance
        file: /test-project/add32.sv
        type-info { source: "reg [31:0]", type ref: (primitive) }
    ) }
    b: { (
        metatype: data/net/var/instance
        file: /test-project/add32.sv
        type-info { source: "reg [31:0]", type ref: (primitive) }
    ) }
    cIn: { (
        metatype: data/net/var/instance
        file: /test-project/add32.sv
        type-info { source: "reg", type ref: (primitive) }
    ) }
    cOut: { (
        metatype: data/net/var/instance
        file: /test-project/add32.sv
        type-info { source: (unknown), type ref: (primitive) }
    ) }
    sum: { (
        metatype: data/net/var/instance
        file: /test-project/add32.sv
        type-info { source: "[31:0]", type ref: (primitive) }
    ) }
  }
}
```

The locations in SymbolTable point to symbols depending on context (looking at Language Server behaviour, `a` points to line `input a, b, Cin;`, `b` points to line `reg [31:0] a, b;`, etc.)

**Expected behavior**

Either a concise pointing to a specified location (e.g. first defining line) or providing a list of all lines defining a given symbol (so that for example Language Server can provide list of locations).

Of course an expected behavior can be established during the discussion.

glatosinski, 2023-03-08:
In the https://github.com/chipsalliance/verible/pull/1678, where we improve support for non-ANSI port declaration, the multi-line definitions of symbols currently result in redefinitions, because all of those lines contribute to the overall typing of the symbol (as described in https://github.com/chipsalliance/verible/pull/1678#issuecomment-1460168315)

hzeller, 2023-03-09:
cc: @fangism

fangism, 2023-03-09:
You could also consider how other similar tools behave, for comparison.

Some ideas for symbol table representation:
* You could add a `struct DeclarationDirectionInfo` next to `DeclarationTypeInfo`, with its own `syntax_origin`.  That would at least let the symbol table record the split declarations.  Obviously for some meta-types, direction is not applicable.

But which definition(s) to expose to the LSP?
The LRM 23.2.2.3 " Rules for determining port kind, data type, and direction" says that both type and direction can be omitted (and implicitly assume defaults).  This means we cannot guarantee that valid code will always have type or direction as a place to anchor a symbol definition.  I think I previously suggested that the non-ANSI port inside the header be a _reference_ to a definition found below inside the module body, but this makes me reconsider.

How would a symbol definition look if the defining symbol was anchored in the non-ANSI ports, the type info pointed to another location, and the direction info pointed to a different location?  What if we allowed the non-ANSI port to initialize an incomplete definition, and allowed type and direction to be attached later?

hzeller, 2023-03-09:
Mostly some random thoughts, for discussions' sake.

So it sounds like we might have multiple properties that a symbol can have and 'accumulate' over time.

Like here, `a` can be refined to be an `input` and then have a 'storage type' `reg` and a type `[31:0]`. So we have multiple slots of information that overall describe a symbol.

To start the discussion we could see if we find a symbol that at first looks like it is about to be re-defined, but then we see that some piece of information is added that we didn't know about that symbol.
First we get the name. Then we see that it is an `input`, later we also are informed about the type. If that 'slot' of information was empty, we can use that to accumulate and update our type-info.

Only when we encounter something that we already have an info for, say previously we got the bit-width of `[31:0]`, and then later again a bit width (same or other), we can complain that that type for `a` was already defined. Or if we see two times that `a` is an `input` (or worse, conflicting `input` and `output`). Once the first time some specific information has been added, it is locked to the symbol - and later we can complain about a specific redefinition part ("type already added in foo.v:42:17:")

(Care must be taken if there is a new scope - so within a `begin` ...`end` we'd prioritize adding a `a` encountered there as a new entry, not something that augments the existing information.)

Now, how to represent that from the language server towards the user ? In the simplest case, we just 'jump to definition' at the first spot we've seen (probably our string-view is keyed rooted in that anyway in some map). We might also consider the _last_ location that we have seen (not modifying the key string-view in the map, but storing within that symbol as 'last definition update'. So we have the original `a` in the port list, and the last `a` where it is `reg [31:0]` enriched.

If we have the last position, maybe that is the one the language server should go to (in our example above, it is actually adds the most important part, the wire-width.). But if the language server can return multiple definitions, then it is simple: we return the first and last time we've seen the symbol in its journey to be fully defined.

PR #1678 (2023-07-04): [LSP] textDocument/definition: Going to ports and module definitions
Fixes #1673
Fixes #1668
Fixes #1783

This PR:

- [X] Adds fix for finding definition of the module when cursor points at symbol definition
- [X] Adds fix for finding definition of ports for module - initial tests added, the solution seems to work.
- [X] Fixes the smoke tests which fail on "redefined" symbols, when symbol definition is fragmented.
- [X] Adds initial multiline goto-definition support to the LanguageServer
- [x] Implements tests for multiline goto-definition
- [x] Implements type checking for split definitions o [... cut]
closed 2023-07-04 by glatosinski