chipsalliance/verible #1786: veriable.flist doesn't work with coc.nvim

zhuzhzh, 2023-03-09:
- vim9 with coc.nvim
 - configure
 
```json
          "verible": {
              "command": "verible-verilog-ls",
              "filetypes": ["systemverilog", "verilog"],
              "settings": {
                  "verilog-trace-parser": "true"
              }
          }
```
-  export VERIBLE_LOGTHRESHOLD=0

I didn't see any special log in `CocInfo`.

`go-to-definition` works if the defination in the same file. It doesn't work if it's in other file.
So `veriable.filelist` doesn't work here.

hzeller, 2023-03-09:
Just checking, the name is `verible.filelist`, not veriable.filelist.

There are two environment variables that might help `VERIBLE_LOGTHRESHOLD=0` and also `VERIBLE_VLOG_DETAIL=1` which expands more detail. Maybe that helps to see what is going on ?

zhuzhzh, 2023-03-09:
@hzeller  sorry, it's my typo. I use verible.filelist in my real environment. I tried `VERIBLE_LOGTHRESHOLD=0` and `VERIBLE_VLOG_DETAIL=1`. but it doesn't output any information in `CocInfo`. I will check the usage of coc.nvim.

hzeller, 2023-03-09:
To verify logging, do you see at least any output ? At least something like the following should be emitted

```
Verible Verilog Language Server built at v0.0-3033-gd35d72e0
... then potentially more logging
```

zhuzhzh, 2023-03-09:
I only see this from coc.nvim. I think I need to find the method to open logging in coc.nvim

```text
2023-03-09T23:21:16.897 INFO (pid:20760) [services] - LanguageClient verible state change: stopped => starting
2023-03-09T23:21:17.707 INFO (pid:20760) [language-client-index] - Language server "languageserver.verible" started with 20792
2023-03-09T23:21:17.730 INFO (pid:20760) [attach] - receive notification: jumpDefinition []
2023-03-09T23:21:17.776 ERROR (pid:20760) [attach] - Error on notification jumpDefinition Error: definition provider not found for current buffer, your language server doesn't support it.
 at Handler.checkProvider (/home/harriszh/.vim/plugged/coc.nvim/build/index.js:83057:17)
 at LocationsHandler.request (/home/harriszh/.vim/plugged/coc.nvim/ build/index.js:79316:22)
at processTicksAndRejections (node:internal/process/task_queues:965) 
at async LocationsHandler.gotoDefinition (/home/harriszh/.vim/plugged/coc.nvim/build/index.js:79358:26)
at async Plugin.cocAction (/home/harriszh/.vim/plugged/coc.nvim/build/index.js:83362:16)
at async EventEmitter.<anonymous> (/home/harriszh/.vim/plugged/coc. nvim/build/index.js:83472:15)
```

zhuzhzh, 2023-03-09:
I found the method to debug lsp in coc.nvim.

`:CocCommand workspace.showOutput` will output the log from lsp.

Then I got the error msg and found out the issue.

My code is in one git reposity.
I put verible.filelist in the exact verilog code (<git_root_dir>/src/v). but verible try to find the verible.filelist in the root directory(<git_root_dir>).

After copying verible.filelist into <git_root_dir>,  it works.

hzeller, 2023-03-09:
Yay.
Yes, editors themselve attempt to find the root of the project and will report it to the language server which then tries to find the file list there. Only if we don't get a file list from the editor, we fall back to backwards search from the location.

We probably need to document that better.