chipsalliance/verible #1801: Can't build on MacOS 13

C1oud555, 2023-03-14:
bazel version:
```
bazel 6.1.0-homebrew
```
clang version:
```
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: arm64-apple-darwin22.3.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```
system version:
```
      System Version: macOS 13.2.1 (22D68)
      Kernel Version: Darwin 22.3.0
```

error info:
```
ERROR: /Users/hyluo/Developer/Builds/verible/common/lsp/BUILD:73:8: Executing genrule //common/lsp:lsp-protocol_gen failed: (Aborted): sandbox-exec failed: error executing command
  (cd /private/var/tmp/_bazel_hyluo/d641108795d6f998f500f2e875fee624/sandbox/darwin-sandbox/256/execroot/com_google_verible && \
  exec env - \
    PATH='/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/Users/hyluo/.emacs.d/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/Users/hyluo/.cargo/bin:/Users/hyluo/.local/bin:/Users/hyluo/Library/Application Support/Coursier/bin:/opt/homebrew/opt/fzf/bin' \
    TMPDIR=/var/folders/9h/bfdqj2157h18dk55x3q41dr40000gn/T/ \
  /usr/bin/sandbox-exec -f /private/var/tmp/_bazel_hyluo/d641108795d6f998f500f2e875fee624/sandbox/darwin-sandbox/256/sandbox.sb /var/tmp/_bazel_hyluo/install/13a283ce36a84067e41018a0f11befbd/process-wrapper '--timeout=0' '--kill_delay=15' '--stats=/private/var/tmp/_bazel_hyluo/d641108795d6f998f500f2e875fee624/sandbox/darwin-sandbox/256/stats.out' /bin/bash -c 'source external/bazel_tools/tools/genrule/genrule-setup.sh; bazel-out/darwin_arm64-opt-exec-2B5CBBC6/bin/common/tools/jcxxgen --json_header='\''"nlohmann/json.hpp"'\'' --class_namespace verible::lsp --output bazel-out/darwin_arm64-opt/bin/common/lsp/lsp-protocol.h common/lsp/lsp-protocol.yaml')
dyld[2206]: symbol not found in flat namespace '_CFRelease'
/bin/bash: line 1:  2206 Abort trap: 6           bazel-out/darwin_arm64-opt-exec-2B5CBBC6/bin/common/tools/jcxxgen --json_header='"nlohmann/json.hpp"' --class_namespace verible::lsp --output bazel-out/darwin_arm64-opt/bin/common/lsp/lsp-protocol.h common/lsp/lsp-protocol.yaml
```

after simple test. Looks like 
```
bazel-out/darwin_arm64-opt-exec-2B5CBBC6/bin/common/tools/jcxxgen
bazel-out/darwin_arm64-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/compiler/protoc_nowkt
```
will raise `dyld[3853]: symbol not found in flat namespace '_CFRelease'` this error

hzeller, 2023-03-14:
There is also a homebrew package, can you try if that would work ?
https://github.com/chipsalliance/homebrew-verible

I am not familiar with common Mac compile errors, so adding the 'help wanted' label; maybe someone recognizes that symptom.

C1oud555, 2023-03-14:
I also tried the `homebrew-verible`. same error occurs.

hzeller, 2023-03-15:
I have heard others were able to compile the homebrew package; so we probably have to wait until someone with Mac compile experience can help us in this thread.

hzeller, 2023-03-15:
Can you compile other sources successfully on your system @C1oud555 ?

C1oud555, 2023-03-15:
@hzeller  seems like all executable under verible will raise this error `symbol not found in flat namespace '_CFRelease'`
it's like some [issue](https://github.com/google/mediapipe/issues/3397#issuecomment-1146456958) with absl link to `CoreFoundation`. 
then I tries add `--linktopt="-framework CoreFoundation"` to bazel build, the error still show up.
And I can compile other cpp project successfully.

hzeller, 2023-03-15:
Interesting. I wonder if we could fix this centrally by changing the `.bazelrc` in the toplevel directory ?

Can you try adding the link options you found in this file, roughly like this ?

```patch
--- a/.bazelrc
+++ b/.bazelrc
@@ -12,7 +12,7 @@ common:linux --cxxopt=-std=c++17 --host_cxxopt=-std=c++17 --client_env=BAZEL_CXX
 
 # https://github.com/abseil/abseil-cpp/issues/848
 # https://github.com/bazelbuild/bazel/issues/4341#issuecomment-758361769
-common:macos --features=-supports_dynamic_linker --cxxopt=-std=c++17 --host_cxxopt=-std=c++17 --client_env=BAZEL_CXXOPTS=-std=c++17
+common:macos --features=-supports_dynamic_linker --host_linkopt="-framework CoreFoundation" --cxxopt=-std=c++17 --host_cxxopt=-std=c++17 --client_env=BAZEL_CXXOPTS=-std=c++17
 
 # Force the use clang-cl on Windows instead of MSVC. MSVC has some issues with
 # the codebase, so we focus the effort for now is to have a Windows Verible
```

C1oud555, 2023-03-15:
After some tests. It's solved by add link option to `.bazelrc` and add `--linktopt="-framework CoreFoundation"` to the bazel build command.
Thanks a lot!

hzeller, 2023-03-15:
Perfect. Now we only need to find out if this will always work on MacOS or if we need to limit that option to Apple Silicon.

hzeller, 2023-03-15:
so is `--linktopt="-framework CoreFoundation"` in the .bazelrc sufficient, or do you need to also provide it on the `bazel build //...` command ? Or does either of them work ? @C1oud555

C1oud555, 2023-03-16:
Both needed. The `--linktopt="-framework CoreFoundation"` in `.bashrc` and the `--linktopt="-framework CoreFoundation"` in the bazel build command.

hzeller, 2023-03-16:
Can you try to patch in https://patch-diff.githubusercontent.com/raw/chipsalliance/verible/pull/1815.diff to your local checkout of Verible and see if you can just `bazel build //...` without any flags and other environment changed ?

Hopefully, this supplies all the necessary linkopts.

C1oud555, 2023-03-16:
It can build without any flags and other environment changed now!

closed 2023-03-20 by C1oud555