chipsalliance/verible #1812: Semicolon, comma and parenthesis alignment issues

Lcr20010506, 2023-03-15:
**Test case**

**Actual output**

```verilog
    wire [MIG_ADDR_WIDTH-1:0] weight_addr_start;
    wire [MIG_ADDR_WIDTH-1:0] weight_addr_end;
    wire [MIG_ADDR_WIDTH-1:0] ddr_addr_start;
    wire [MIG_ADDR_WIDTH-1:0] ddr_addr_end;
    wire [              31:0] remove;
    wire [              31:0] offset;
    wire [              31:0] weight_len;
    wire [              31:0] neuron_num_in_layer;
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```verilog
    wire [MIG_ADDR_WIDTH-1:0] weight_addr_start     ;
    wire [MIG_ADDR_WIDTH-1:0] weight_addr_end       ;
    wire [MIG_ADDR_WIDTH-1:0] ddr_addr_start        ;
    wire [MIG_ADDR_WIDTH-1:0] ddr_addr_end          ;
    wire [              31:0] remove                ;
    wire [              31:0] offset                ;
    wire [              31:0] weight_len            ;
    wire [              31:0] neuron_num_in_layer   ;
```
the same with `)` and `,`
**Expected or suggested output**
```verilog
    generate_ddr_addr #(
        .FLOAT_LEN        (FLOAT_LEN        ),
        .MIG_DATA_WIDTH   (MIG_DATA_WIDTH   ),
        .MIG_ADDR_WIDTH   (MIG_ADDR_WIDTH   ),
        .LAYER0_NEURON_NUM(LAYER0_NEURON_NUM),
        .LAYER1_NEURON_NUM(LAYER1_NEURON_NUM),
        .LAYER2_NEURON_NUM(LAYER2_NEURON_NUM),
        .LAYER3_NEURON_NUM(LAYER3_NEURON_NUM),
        .LAYER4_NEURON_NUM(LAYER4_NEURON_NUM),
        .MAX_WEIGHT_NUM   (MAX_WEIGHT_NUM   )
    ) 
```
This may seem strange, but in some cases (such as instantiating modules) it works quite well, do we have the relevant configuration parameters to support this effect?

hzeller, 2023-03-15:
I have not seen a request to align semicolons yet, but might be something that could probably be added and made configurable.

(Configuration of options in general can be done with flags. There are often different choices per flag.
https://github.com/chipsalliance/verible/tree/master/verilog/tools/formatter#systemverilog-formatter
)

As for your observation 'sometimes it works': there is a heuristic in the formatter that attempts to determine what desired indentation is wanted (you see this in the flags if the style is set to `infer`). That typically means that the formatter looks at the surroundings (e.g. an unusual indentation of more than one space) to infer what particular formatting was "meant". That of course does not work always, in particular if you feed a differently formatted file.

So if there is a particular style that you prefer for certain constructs, you can lock them in and change the value from `infer` to your desired style.

zjulijinxin, 2023-07-21:
> **Test case**
>
> **Actual output**
>
> ```verilog
>     wire [MIG_ADDR_WIDTH-1:0] weight_addr_start;
>     wire [MIG_ADDR_WIDTH-1:0] weight_addr_end;
>     wire [MIG_ADDR_WIDTH-1:0] ddr_addr_start;
>     wire [...]

Hi, have you found the configuration for aligning ")" and "," yet? I am also looking for how to align these symbols, as aligning them will make them look cleaner.