chipsalliance/verible #1821: [Build Error] error: 'transform_reduce' is not a member of 'std'

jagjordi, 2023-03-20:
I get this error when trying to build verible

OS: RHEL 8
GCC version 9.2.1
 
I use SCL gcc-toolset-9 to get GCC9 into RHEL8, it normally works well with other programs 

```
[jordi@mozzarella verible]$ bazel run -c opt --//bazel:use_local_flex_bison //:install -- -s /usr/local/bin
INFO: Build option --//bazel:use_local_flex_bison has changed, discarding analysis cache.
INFO: Analyzed target //:install (0 packages loaded, 3230 targets configured).
INFO: Found 1 target...
INFO: From Executing genrule //common/analysis:command_file_lex:
common/analysis/command_file.lex:77: warning, rule cannot be matched
ERROR: /home/jordi/Downloads/verible/common/util/BUILD:259:11: Compiling common/util/tree_operations.cc failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 24 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from common/util/tree_operations.cc:15:
./common/util/tree_operations.h: In function 'void verible::FlattenOnce(T&)':
./common/util/tree_operations.h:728:40: error: 'transform_reduce' is not a member of 'std'
   const int grandchildren_count = std::transform_reduce(
                                        ^~~~~~~~~~~~~~~~
./common/util/tree_operations.h:728:40: note: suggested alternative: 'transform'
   const int grandchildren_count = std::transform_reduce(
                                        ^~~~~~~~~~~~~~~~
                                        transform
./common/util/tree_operations.h: In function 'void verible::FlattenOnlyChildrenWithChildren(T&, std::vector<long unsigned int>*)':
./common/util/tree_operations.h:760:39: error: 'transform_reduce' is not a member of 'std'
   const int new_children_count = std::transform_reduce(
                                       ^~~~~~~~~~~~~~~~
./common/util/tree_operations.h:760:39: note: suggested alternative: 'transform'
   const int new_children_count = std::transform_reduce(
                                       ^~~~~~~~~~~~~~~~
                                       transform
Target //:install failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 3.237s, Critical Path: 2.42s
INFO: 96 processes: 49 internal, 47 linux-sandbox.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully
```

hzeller, 2023-03-22:
Verible uses c++17; (and the [std::transform_reduce](https://en.cppreference.com/w/cpp/algorithm/transform_reduce) should be defined in the `#include <numeric>` standard header).
It could be that  the standard library that came with your gcc 9 did not support all C++17 features yet ?

We regularly build the [releases](https://github.com/chipsalliance/verible/releases) in the CI with [gcc-9](https://github.com/chipsalliance/verible/actions/runs/4481149094/jobs/7877455717#step:5:151), so in general gcc-9 seems to be capable, so I suspect it might just be a standard header issue.

Is it possible for you to install a newer than stock gcc on your machine ?

jagjordi, 2023-05-06:
Sorry for the late reply.
I have now tried with GCC 9.2.1 and I am getting the same errors

hzeller, 2023-07-22:
This does sound like you compiler set-up is not ready for c++17 yet. I can reproduce your error if I test with a very ancient gcc 8.5 compiler.

It does work for the current gcc-9 (9.5.0) for me.

So gcc 9 was probably right at the border where some c++17 features/library functions were not supported yet.

Since gcc9 is really old, Is it possible to install a slightly newer versions of gcc and standard c++ library on your system ?
https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8/html/developing_c_and_cpp_applications_in_rhel_8/additional-toolsets-for-development_developing-applications

hzeller, 2023-07-22:
RedHat also poits out that the version 9 of the compiler [only is experimental for c++17](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8/html/developing_c_and_cpp_applications_in_rhel_8/additional-toolsets-for-development_developing-applications#:~:text=C%2B%2B17%2C,cannot%20be%20guaranteed) ... which is the c++ standard we need.

So upgrading to toolchain >= 10 is a good choice.

closed 2023-07-22 by hzeller