chipsalliance/verible #1832: Option to have alignment only apply to blocks of code

patrickrst, 2023-03-24:
Hi, here is an idea for an improvement for the formatter. In some codebases, there is sometime very long variable/parameter names, which makes the aligned output a bit harder to read:

**Actual output**

```systemverilog
module test;
  logic [                        WIDTH - 1:0] some_long_name1[SOME_LONG_PARAM-1:0];
  logic [                        WIDTH - 1:0] some_long_name2[SOME_LONG_PARAM-1:0];
  logic [                        WIDTH - 1:0] some_long_name3[SOME_LONG_PARAM-1:0];

  logic                                       addr           [              N-1:0];

  logic [my_pkg::ANOTHER_LONG_PARAM_NAME-1:0] data           [              N-1:0];
endmodule
```

**Suggested output**

```systemverilog
module test;
  logic [WIDTH - 1:0] some_long_name1[SOME_LONG_PARAM-1:0];
  logic [WIDTH - 1:0] some_long_name2[SOME_LONG_PARAM-1:0];
  logic [WIDTH - 1:0] some_long_name3[SOME_LONG_PARAM-1:0];

  logic addr[N-1:0];

  logic [my_pkg::ANOTHER_LONG_PARAM_NAME-1:0] data[N-1:0];
endmodule
```

It would be nice to have the alignment be done by block of code. In the same way that is done in ports declaration, where a newline act as a separator.

I don't know what separator would be best to represent a new block. In this example the blocks are separated with a newline, but it could be two newlines or a line with a comment that acts as a separator.

The same behaviour is also observed with assignments, they are all aligned even if there is a newline separating them.

shareefj, 2023-08-08:
I was about to post something similar but thought I'd add to this request.

I've just tried the formatter again and see that setting `--assign_statement_alignment=align` seems to align all assignments in the file.  What I'd like to see is that blocks of code separated by whitespace are treated separately.  I don't want to see lots of whitespace inserted in separate blocks of code.

What I'd like to see is:

```verilog
assign x   = one;
assign y   = two;
assign xyz = three;

assign some_long_name   = four;
assign some_longer_name = five;
```

What I actually get is:

```verilog
assign x                = one;
assign y                = two;
assign xyz              = three;

assign some_long_name   = four;
assign some_longer_name = five;
```

bbo49, 2025-03-12:
As a workaround, I currently insert  `` `define`` macros as a separator, to break declarations in different formatting sections.
The macro name can be chosen arbitrary. 

```verilog
module test;
  `define _VERIBLE_FORMAT_REGION
  logic [WIDTH - 1:0] some_long_name1[SOME_LONG_PARAM-1:0];
  logic [WIDTH - 1:0] some_long_name2[SOME_LONG_PARAM-1:0];
  logic [WIDTH - 1:0] some_long_name3[SOME_LONG_PARAM-1:0];

  `define _VERIBLE_FORMAT_REGION
  logic addr[N-1:0];

  `define _VERIBLE_FORMAT_REGION
  logic [my_pkg::ANOTHER_LONG_PARAM_NAME-1:0] data[N-1:0];
endmodule
````

kbrunham-intel, 2026-09-22:
@hzeller This looks like a duplicate of #814. I would suggest closing this issue in favor of that one so we keep a single tracking ticket.