chipsalliance/verible #1833: Couldn't start client Verible Language Server - Verible VS Code plug in

miguelcruz97, 2023-03-24:
I installed the plug-in in VS Code:
[image]

After installation, following message appears:

[image]

Output:
[image]

I have install verible-verilog-ls and I can execute it from the command line:
[image]

I tried to check if further configuration was needed but the plug in does not support any configuration. Binaries are added to the PATH variable.

hzeller, 2023-03-24:
If you go to the 'settings' of the plug-in (the little gear :gear:-symbol at the bottom right corner in the Verible extension listing). you can set the path to the `verible-verilog-ls` binary. Maybe that is pointing to some other place than your binary is installed at ?

miguelcruz97, 2023-03-24:
You are right! I was not able to see that option because I was checking the User window (where verible was not installed) instead of the Remote one. Thank you, solved!

closed 2023-03-24 by miguelcruz97