chipsalliance/verible #1843: Language server crashes when trying to format incorrect code

arunkmv, 2023-03-29:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ?
Emacs
 - IDE version:
28.1
 - What other SystemVerilog plugins are active alongside ?
verilog-mode
-----

**What activity failed**
Formatting

**Test case**

- *Case 1*
Trailing comma in parameter list.
```systemverilog
module test_module #(
    parameter int SomeParam = 8,
    parameter int SomeOtherParam = 16,
) (
    input  logic a,
    input  logic b,
    output logic c
);
  assign c = a & b;
endmodule
```

- *Case 2*
Trailing comma in port list.
```systemverilog
module test_module #(
    parameter int SomeParam = 8,
    parameter int SomeOtherParam = 16
) (
    input  logic a,
    input  logic b,
    output logic c,
);
  assign c = a & b;
endmodule
```

- *Case 3*
Missing semicolon.
```systemverilog
module test_module #(
    parameter int SomeParam = 8,
    parameter int SomeOtherParam = 16
) (
    input  logic a,
    input  logic b,
    output logic c
);
  assign c = a & b
endmodule
```

On trying to format this code using verible-verilog-ls, the language server crashes. Stacktrace is same for all three cases.

<details>
  <summary>Crash Stacktrace</summary>

```
Verible Verilog Language Server built at v0.0-3110-gbb28b0b4
F0329 12:43:20.088173 2152954 tree_unwrapper.cc:78] Check failed: 'text_structure_view_.SyntaxTree()' Must be non-null
*** Check failure stack trace: ***
    @           0x46ce24  absl::log_internal::LogMessage::SendToLog()
    @           0x46cb6b  absl::log_internal::LogMessage::Flush()
    @           0x46d22f  absl::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x46c140  absl::log_internal::DieBecauseNull()
    @           0x3f4d4f  verible::TreeUnwrapper::Unwrap()
    @           0x3b76d6  verilog::formatter::Formatter::Format()
    @           0x3b7105  verilog::formatter::FormatVerilog()
    @           0x30626a  verilog::FormatRange()
    @           0x30336c  std::_Function_handler<>::_M_invoke()
    @           0x45a953  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x45969a  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x467d73  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x467a21  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x467918  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x2fb56e  verilog::VerilogLanguageServer::Run()
    @           0x2f8139  main
    @     0x7fbd224f3d85  __libc_start_main
*** SIGABRT received at time=1680086600 on cpu 15 ***
PC: @     0x7fbd22507aff  (unknown)  raise
    @           0x46eeee        208  absl::WriteFailureInfo()
    @           0x46ec70         64  absl::AbslFailureSignalHandler()
    @     0x7fbd22891cf0  (unknown)  (unknown)
    @           0x46ce84         32  absl::log_internal::LogMessage::SendToLog()
    @           0x46cb6b         80  absl::log_internal::LogMessage::Flush()
    @           0x46d22f         32  absl::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x46c140        240  absl::log_internal::DieBecauseNull()
    @           0x3f4d4f         64  verible::TreeUnwrapper::Unwrap()
    @           0x3b76d6        944  verilog::formatter::Formatter::Format()
    @           0x3b7105        656  verilog::formatter::FormatVerilog()
    @           0x30626a        304  verilog::FormatRange()
    @           0x30336c        144  std::_Function_handler<>::_M_invoke()
    @           0x45a953        144  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x45969a        256  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x467d73        256  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x467a21        240  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x467918         32  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x2fb56e         64  verilog::VerilogLanguageServer::Run()
    @           0x2f8139        672  main
    @     0x7fbd224f3d85  (unknown)  __libc_start_main
    @ 0x5541f689495641d7  (unknown)  (unknown)
Verible Verilog Language Server built at v0.0-3110-gbb28b0b4

Process verible-ls stderr<1> finished

Process verible-ls stderr finished
```
</details>

**Expected or suggested output**

While it is not expected to format incorrect code, the language server should not crash. It must just not provide formatting changes.

HendrikMennen, 2023-04-15:
Noticed this too. I had to disable formatting in my IDE because of this

IEncinas10, 2023-04-16:
I gave it a try and [this](https://github.com/IEncinas10/verible/commit/7273bb4f46244c8360c52a384968b732cab1bee4) minor change seems to fix the crash. I just checked with my local editor and stops crashing.

IEncinas10, 2023-04-19:
@hzeller Please give this a look, if I haven't missed anything I'll do the PR. 

* Pinging because it looks important enough, please let me know if it's ok with you :)

closed 2023-04-24 by hzeller