chipsalliance/verible #1856: Build fails

yurivict, 2023-04-05:
**Describe the bug**

Build fails with this error:

```
$ bazel build -c opt //...
Extracting Bazel installation...
Starting local Bazel server and connecting to it...
WARNING: Download from https://mirror.bazel.build/github.com/westes/flex/releases/download/v2.6.4/flex-2.6.4.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
INFO: Analyzed 1893 targets (151 packages loaded, 6898 targets configured).
INFO: Found 1893 targets...
ERROR: /usr/home/yuri/.cache/bazel/_bazel_yuri/c8363c06afa4c26207790ac1c5fde5cb/external/com_google_absl/absl/hash/BUILD.bazel:151:11: Compiling absl/hash/internal/low_level_hash.cc failed: (Exit 1): clang failed: error executing command /usr/bin/clang -U_FORTIFY_SOURCE '-D_FORTIFY_SOURCE=1' -fstack-protector -Wall -fno-omit-frame-pointer -g0 -O2 -DNDEBUG -ffunction-sections -fdata-sections '-std=c++0x' -MD -MF ... (remaining 32 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from external/com_google_absl/absl/hash/internal/low_level_hash.cc:15:
In file included from external/com_google_absl/absl/hash/internal/low_level_hash.h:31:
In file included from external/com_google_absl/absl/base/config.h:86:
external/com_google_absl/absl/base/policy_checks.h:79:2: error: "C++ versions less than C++14 are not supported."
#error "C++ versions less than C++14 are not supported."
 ^
1 error generated.
ERROR: /usr/home/yuri/.cache/bazel/_bazel_yuri/c8363c06afa4c26207790ac1c5fde5cb/external/com_google_protobuf/src/google/protobuf/BUILD.bazel:142:11 Middleman _middlemen/@com_Ugoogle_Uprotobuf_S_Ssrc_Sgoogle_Sprotobuf_Cwkt_Ucc_Uproto-BazelCppSemantics_build_arch_freebsd-opt-exec-2B5CBBC6 failed: (Exit 1): clang failed: error executing command /usr/bin/clang -U_FORTIFY_SOURCE '-D_FORTIFY_SOURCE=1' -fstack-protector -Wall -fno-omit-frame-pointer -g0 -O2 -DNDEBUG -ffunction-sections -fdata-sections '-std=c++0x' -MD -MF ... (remaining 32 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
INFO: Elapsed time: 23.443s, Critical Path: 0.51s
INFO: 37 processes: 36 internal, 1 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
```

Is there any way to build with cmake?

clang-14
bazel-5.3.0
FreeBSD 13.1

tgorochowik, 2023-04-11:
We do have a clang-12 build in the CI (sample build here: https://github.com/chipsalliance/verible/actions/runs/4602833394/jobs/8132234894)

We need to check whether the failure you see is caused by newer clang, freebsd or a combination of factors.

yurivict, 2023-04-11:
It generally depends on compiler + c++ library combination.

hzeller, 2023-04-11:
We do have the needed c++17 flags in  the .bazelrc, but it is separated by platforms, as Windows requires its own sets of fields; here it is explicitly set for 'common:linux'
https://github.com/chipsalliance/verible/blob/master/.bazelrc#L11

I suspect there might be something like 'common:bsd' or similar where this can can be made work ? Ideally, maybe a more generic term such as `common:unix` would be good. Something to extract from the bazel documentation and then fixed in the .bazelrc.

hzeller, 2023-04-12:
Can you try again from head ? Settings should now work for freebsd and openbsd.

hzeller, 2023-04-12:
Can you also add `--host_linkopt=-lm` ?

yurivict, 2023-04-12:
Adding ```--linkopt="-lm" --host_linkopt=-lm``` helped - now it finishes successfully.

hzeller, 2023-04-12:
Alright, can you try again compiling from head ?

yurivict, 2023-04-12:
It builds now, thank you.

However, it downloads files during build. This isn't allowed from the package builder.

Particularly, it downloads more files from bazel scripts in the protobuf tarball.
Do you know if there is a way to use pre-installed protobuf instead?

hzeller, 2023-04-17:
yes, I am sorry, the vendored dependencies in bazel are a common headache while packaging, as bazel insists on fetching it themselve.

I think there is a way to do the vendored dependencies from local places, but I have not explored that too much.

There is a way to separate the 'fetching dependencies' (which needs internet access) and 'building' (which does not). The following illustrates how that can work.

It is not optimal, but it can be a starting point for the packaging

```bash
#!/usr/bin/env bash

# Use BASE_DIR if set, otherwise TMPDIR
export TMPDIR=${TMPDIR:-/tmp}
BASE_DIR=${BASE_DIR:-$TMPDIR}

# Use local flex and bison
ADDITIONAL_BAZEL_OPTIONS="--//bazel:use_local_flex_bison"

# Where we want the final installation to be
PREFIX=${PREFIX:-${BASE_DIR}/install}

# Packaging directory: contains the source and all dependencies
PACKAGE_DIR="${BASE_DIR}/verible-package"

mkdir -p "${PACKAGE_DIR}"
git clone https://github.com/chipsalliance/verible "${PACKAGE_DIR}/src"

cd "${PACKAGE_DIR}/src"

# Force fetching all the dependencies. This should work with
# bazel fetch //..., but because of
# https://github.com/bazelbuild/bazel/issues/13847 this is not working yet.
# But the trick forcing to fetch everything by querying the deps works:
bazel --output_base="${PACKAGE_DIR}/bazel" cquery ${ADDITIONAL_BAZEL_OPTIONS} "deps(//...)" > /dev/null

########################################################
#----- Internet needed above, but not below anymore ---
########################################################

cd "${PACKAGE_DIR}/src"
bazel --output_base="${PACKAGE_DIR}/bazel" build ${ADDITIONAL_BAZEL_OPTIONS} -c opt --fetch=false :install-binaries

# Install binaries using the install tool. I think that might use some
# gnu-install flags, so hopefully it also works on BSD; if not, please report,
# then let's fix that.
bazel --output_base="${PACKAGE_DIR}/bazel" run ${ADDITIONAL_BAZEL_OPTIONS} -c opt --fetch=false :install -- "${PREFIX}/bin"

echo "Binaries are now in ${PREFIX}/bin"
ls -l "${PREFIX}/bin"
```