chipsalliance/verible #1857: fatal error: 'stdio_ext.h' file not found

yurivict, 2023-04-05:
**Describe the bug**
```
$ CXXFLAGS="-sta=c++20" bazel build -c opt //... --cxxopt=-std=c++20
INFO: Build option --cxxopt has changed, discarding analysis cache.
WARNING: Download from https://mirror.bazel.build/github.com/westes/flex/releases/download/v2.6.4/flex-2.6.4.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
INFO: Analyzed 1893 targets (0 packages loaded, 6898 targets configured).
INFO: Found 1893 targets...
ERROR: /usr/home/yuri/.cache/bazel/_bazel_yuri/c8363c06afa4c26207790ac1c5fde5cb/external/m4_v1.4.18/gnulib/BUILD.bazel:205:11: Compiling gnulib/lib/close-stream.c [for host] failed: (Exit 1): clang failed: error executing command /usr/bin/clang -U_FORTIFY_SOURCE '-D_FORTIFY_SOURCE=1' -fstack-protector -Wall -fno-omit-frame-pointer -g0 -O2 -DNDEBUG -ffunction-sections -fdata-sections -MD -MF ... (remaining 22 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from external/m4_v1.4.18/gnulib/lib/close-stream.c:25:
external/m4_v1.4.18/gnulib/lib/fpending.h:24:11: fatal error: 'stdio_ext.h' file not found
# include <stdio_ext.h>
          ^~~~~~~~~~~~~
1 error generated.
INFO: Elapsed time: 92.447s, Critical Path: 25.33s
INFO: 503 processes: 232 internal, 271 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
```

Where is ```stdio_ext.h``` described?

clang-14
bazel-5.3.0
FreeBSD 13.1

hzeller, 2023-04-11:
m4 is built as part of building flex and bison and that is not always working well on all systems. The error you see happens while m4 is compiled.

If you have flex and bison installed locally, you can use the following option to just use the local system-provided tools:

```
bazel build -c opt  --//bazel:use_local_flex_bison //...
```

tgorochowik, 2023-06-01:
@yurivict does the suggestion above help you?

closed 2023-07-22 by hzeller