chipsalliance/verible #1874: Use colored patch output in diff ?

hzeller, 2023-04-19:
Since #1866 we have a nicely colored patch.

We have similar code to output diffs in [diff.cc](https://github.com/chipsalliance/verible/blob/master/common/strings/diff.cc#L158) which does not do any coloring.

It might be a good code-quality work to refactor the code in patch and diff to use the same code to output diffs.

hzeller, 2023-04-19:
The unified diff output is for instance used in the output of showing alternatives for lint output.

Say we have the following file `/tmp/undersized_binary_literal.sv`

```systemverilog
module undersized_binary_literal;
  localparam logic [3:0] Foo = 4'b1;
endmodule
```

... then running the linter shows the alternatives as a unified diff. Would be cool if they would also be colored:
```
$ verible-verilog-lint --autofix=patch-interactive /tmp/undersized_binary_literal.sv 
/tmp/undersized_binary_literal.sv:2:35: Binary literal 4'b1 has less digits than expected for 4 bits. [Style: number-literals] [undersized-binary-literal]
[ 1. Alternative Left-expand leading zeroes ]
@@x -1,3 +1,3 @@
 module undersized_binary_literal;
-  localparam logic [3:0] Foo = 4'b1;
+  localparam logic [3:0] Foo = 4'b0001;
 endmodule
[ 2. Alternative Replace with decimal ]
@@x -1,3 +1,3 @@
 module undersized_binary_literal;
-  localparam logic [3:0] Foo = 4'b1;
+  localparam logic [3:0] Foo = 4'd1;
 endmodule
[ 3. Alternative Adjust width to inferred width ]
@@x -1,3 +1,3 @@
 module undersized_binary_literal;
-  localparam logic [3:0] Foo = 4'b1;
+  localparam logic [3:0] Foo = 1'b1;
 endmodule
Autofix is available. Apply? [1,2,3,y,n,a,d,A,D,p,P,?] 
```

hzeller, 2023-04-19:
@IEncinas10 is this something that would interest you ? It essentially would be some refactoring to improve the diff output everywhere.

IEncinas10, 2023-04-20:
Sure! Missed this part, as I only knew the script for interactive formatting, so I thought maybe patch.cc was it.

hzeller, 2023-04-20:
The code came together at different times, so this is why there is somewhat duplicate efforts. Might be worthwhile using that opportunity to see if they can be refactored together.

IEncinas10, 2023-04-21:
### Q1: What do you think about adding some highlighting to the linting rule? 

We have to do something different because the line above will be highlighted with "bold", the line below with "inverse" so leaving it in plain text makes it hard to see, and as it is the rule the user is breaking it makes sense for me to highlight it somehow. 

[image]

[image]

Quick reference for alternatives:

- https://en.wikipedia.org/wiki/ANSI_escape_code#3-bit_and_4-bit

### Q2: The way of refactoring together I see is by adding new constructors to patch.h classes 

MarkedLine, HunkIndices, HunkHeader that come from processed data instead of text. 

Don't know if this is what you had in mind and I see reasons for disliking it, I'll let you be the judge :). I'll push in my fork and edit this comment so you can see what I'm saying exactly, or if you prefer I could do a draft-PR.

hzeller, 2023-04-21:
A1: I think we should not overdo the coloring. The regular output of `filename:<line>:<col>` is the expected regular output of the linter looks exactly as the expected output of the linter, just that we now added a bit more visible UI.

A2: Let's start with a draft pull request and discuss the details there. I think in general if we can end up with a higher-level representation of how edits look like and passed around (beyond simple text), i.e. use `diff::Edit`, the better and more universally we can deal with edit scripts in general. My instinct would actually be to represent patches as `diff::Edit` scripts, and then use that as base to render patches, but I am not sure if this will preserve everything we need to represent a `Hunk`. It might be a multi-step refeactoring. So maybe start out with: what is the minimum you'd need to do to have both of these use the same code to render a colorful patch; then you'll have enough familarity with the code to then suggest further refactoring steps.