chipsalliance/verible #1883: verible-verilog-format fails verification with macro

nathanbeckmann, 2023-04-22:
**Test case**

```
module test

    (
     input logic x,
     output logic y
     );

    always_comb begin
        if (`IS_ZERO(x)) begin
            y <= x;
        end
    end

endmodule
```

Include any options or configuration used.

**Actual output**

```
✗ verible-verilog-format macro/mem/behavioral/test.sv 
module test

    (
     input logic x,
     output logic y
     );

    always_comb begin
        if (`IS_ZERO(x)) begin
            y <= x;
        end
    end

endmodule
macro/mem/behavioral/test.sv: Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#712: ")")
First mismatched token [19]: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#712: ")")
; problematic formatter output is
module test (
    input  logic x,
    output logic y
);

  always_comb begin
    if (
        `IS_ZERO(x)
        ) begin
      y <= x;
    end
  end

endmodule
<<EOF>>
```

**Expected or suggested output**

The macro should probably not have newlines surrounding it, and verification should succeed because (as far as I can tell) the output is equivalent to the input.

nathanbeckmann, 2023-04-22:
Observations
 * Removing the backtick from the input fixes the problem.
 * Moving the macro out of the condition and into the assignment fixes the problem. That is, ```  always_comb begin
    if (x) begin
      y <= `IS_ZERO(x);
    end
  end```

nathanbeckmann, 2023-04-22:
I am seeing a possibly related failure on macros in declarations:

```
✗ verible-verilog-format macro/mem/behavioral/test.sv
module test

    (
     input logic x,
     output logic y
     );

    logic [`WIDTH(x):0] z;

    always_comb begin
        if (x) begin
            y <= x;
        end
    end

endmodule
macro/mem/behavioral/test.sv: Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#712: ")")
First mismatched token [17]: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#712: ")")
; problematic formatter output is
module test (
    input  logic x,
    output logic y
);

  logic [
  `WIDTH(x)
  :0] z;

  always_comb begin
    if (x) begin
      y <= x;
    end
  end

endmodule
<<EOF>>
```

If I remove the parameter from the macro, then verification succeeds:

```
✗ verible-verilog-format macro/mem/behavioral/test.sv
module test (
    input  logic x,
    output logic y
);

  logic [`WIDTH:0] z;

  always_comb begin
    if (x) begin
      y <= x;
    end
  end

endmodule
```

Applying the same principle to the earlier test case, it appears that macros with arguments are the problem:

```
✗ verible-verilog-format macro/mem/behavioral/test.sv
module test (
    input  logic x,
    output logic y
);

  always_comb begin
    if (`CONDITION) begin
      y <= x;
    end
  end

endmodule
```

nathanbeckmann, 2023-04-22:
FWIW, here are the flags I'm using:

```
--column_limit=100
--indentation_spaces=4
--assignment_statement_alignment=flush-left
--case_items_alignment=flush-left
--class_member_variable_alignment=flush-left
--compact_indexing_and_selections=true
--distribution_items_alignment=flush-left
--enum_assignment_statement_alignment=flush-left
--expand_coverpoints=false
--formal_parameters_alignment=align
--formal_parameters_indentation=wrap
--module_net_variable_alignment=flush-left
--named_parameter_alignment=align
--named_parameter_indentation=wrap
--named_port_alignment=flush-left
--named_port_indentation=wrap
--port_declarations_alignment=align
--port_declarations_indentation=wrap
--port_declarations_right_align_packed_dimensions=false
--port_declarations_right_align_unpacked_dimensions=false
--struct_union_members_alignment=flush-left
--try_wrap_long_lines=true
--failsafe_success=false
```

hzeller, 2023-04-22:
Thanks for the report with detailed examples and disection of the relevant parts!

This looks like a duplicate of #1524 that we have lined up on the issues to next work on.

hzeller, 2023-04-25:
Since this is similar to #1524 , assigning to you @jbylicki

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

`if (\`IS_ZERO(x))` no longer fails lex-verify (`MacroCallCloseToEndLine`).