chipsalliance/verible #1889: Instantiations Wrapped with One Port

bkueffle, 2023-04-24:
Behavior around module instantiation wrapping doesn't seem desirable. Whenever only one port and/or parameter is given, the instantiation is done on one line (as long as it fits on the line). If 2 ports or 2 parameters are given, then the instantiation style is wrapped, even if it does fit on one line.

**Test case**

```systemverilog
  module_name #(
      .Param1(Param1)
  ) only_one_param_or_port (
      .port_i(a_i)
  );

  module_name #(
      .Param1(Param1),
      .Param2(Param2)
  ) two_or_more_param_or_ports (
      .port_i(a_i)
  );

  module_name fits_on_one_line (
      .port_i(a_i),
      .port_o(b_o)
  );
```

**Actual output**

```systemverilog
  module_name #(.Param1(Param1)) only_one_param_or_port (.port_i(a_i));

  module_name #(
      .Param1(Param1),
      .Param2(Param2)
  ) two_or_more_param_or_ports (
      .port_i(a_i)
  );

  module_name fits_on_one_line (
      .port_i(a_i),
      .port_o(b_o)
  );
```

Include any possible diagnostic messages from the formatter.

**The output I desire**

I always want to wrap my instantiations, even if it fits on one line.

```systemverilog
  module_name #(
      .Param1(Param1)
  ) only_one_param_or_port (
      .port_i(a_i)
  );

  module_name #(
      .Param1(Param1),
      .Param2(Param2)
  ) two_or_more_param_or_ports (
      .port_i(a_i)
  );

  module_name fits_on_one_line (
      .port_i(a_i),
      .port_o(b_o)
  );
```

**Another Output Option that Some May Prefer**

Some may prefer making the instantiation fit on one line if it is <= the character limit.
```systemverilog
  module_name fits_on_one_line (.port_i(a_i), .port_o(b_o));
```

hzeller, 2023-04-24:
Thanks for the report. Yes indeed, these things should be more configurable. So three choices would be good:  multi-line always, single-line iff one parameter, single-line if all parameters fit within line length.

@jbylicki Since you're on a roll currently with formatter updates, is this something that interests you ?

bkueffle, 2023-04-24:
Yeah to prevent bloat I think just two options would suffice:

Multi-line always.
Single-line if all parameters & ports fit within line length.

single-line iff one parameter and/or port seems like a bug/side effect of some other rule. I don't know any style that would use this.