chipsalliance/verible #1893: syntax error at token "`elsif"

skazarynau1, 2023-04-26:
**Test case**

```systemverilog
module m1  ;

`ifdef R1
    r1
`elsif R2
    r2
`else
    r3
`endif
        r (
            .i_clk           (i_clk                         ));
endmodule  
```

```
verible-verilog-format --inplace=true --compact_indexing_and_selections=false
```

**Actual output**

```
./example.sv: ./example.sv:6:1-6: syntax error at token "`elsif"
./example.sv:13:1-9: syntax error at token "endmodule"
```

```
verible-verilog-format --version
v0.0-3222-gb19cdf44
Commit  2023-04-25
Built   2023-04-26T09:02:38Z
```

DivyanshSingh12, 2023-04-27:
where can i edit it

hzeller, 2023-04-27:
Formatting with multiple preporocessing branches is done in a way that it just considers all branches concatenated; however in this case it would of course result in an invalid SystemVerilog.

Ideally, the formatter should take each branch into account individually. This is something that we consider working on in the coming weeks/months, so stay tuned.

hzeller, 2023-04-27:
Assigning to Tom to triage. Maybe also look through the bug tracker and combine all the similar problems into one meta-bug.
We might want to skip the full parse for the formatter and directly only operate on the token stream if possible.