chipsalliance/verible #1894: Mismatched token enums (macro with argument)

skazarynau1, 2023-04-26:
**Test case**

```systemverilog
module m1;
    task automatic t1 ();
        t2(`R1(1) + 8);
    endtask : t1
endmodule
```
Options:
```
verible-verilog-format --inplace=true --compact_indexing_and_selections=false
```

**Actual output**

```systemverilog
./example.sv: Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#712: ")")
First mismatched token [14]: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#712: ")")
; problematic formatter output is

module m1;
  task automatic t1();
    t2(
    `R1(1)
    + 8);
  endtask : t1
endmodule
<<EOF>>
```

```
verible-verilog-format --version
v0.0-3222-gb19cdf44
Commit  2023-04-25
Built   2023-04-26T09:02:38Z
```

hzeller, 2023-05-05:
After recent changes, this now results now in:
@jbylicki can you have a look ?

```
module m1;
    task automatic t1 ();
        t2(`R1(1)+8);
    endtask:t1
endmodule
/tmp/a.sv: Re-formatted text does not match formatted text; formatting failed to converge!  Please file a bug.
========== Original: --lines: ==========
module m1;
    task automatic t1 ();
        t2(`R1(1)+8);
    endtask:t1
endmodule
============== Formatted: ==============
module m1;
  task automatic t1();
    t2(
    `R1(1)
    + 8);
  endtask : t1
endmodule
============= Re-formatted: ============
module m1;
  task automatic t1();
    t2(`R1(1)
       + 8);
  endtask : t1
endmodule
============== Diffs are: ==============
 module m1;
   task automatic t1();
-    t2(
-    `R1(1)
-    + 8);
+    t2(`R1(1)
+       + 8);
   endtask : t1
 endmodule
; problematic formatter output is
module m1;
  task automatic t1();
    t2(
    `R1(1)
    + 8);
  endtask : t1
endmodule
<<EOF>>
```

hzeller, 2023-05-05:
Related #1524

jbylicki, 2023-05-11:
I'm working on it

PR #1917 (2023-07-04): Fix macro formatting divergence and splitting issues
This PR removes a unnecessary check when dealing with kReferenceCallBase where the r_paren token should be added regardless of partition policy (as the style dictates that the bracket is always connected to the last token)

Fixes #1894 
Fixes #1924