chipsalliance/verible #1904: Auto-push to chipsalliance/homebrew-verible from CI

hzeller, 2023-05-02:
We have a https://github.com/chipsalliance/homebrew-verible repository ( #758 ) that contains a MacOS installable package description.

Right now, it is only updated manually, but since we have all information in our verible CI, we should be able to automatically commit to the homebrew-verible repository with the updated version number and hash.

 [This would be a typical change.](https://github.com/chipsalliance/homebrew-verible/commit/835ce1c0f6f1348559d86c211d9ec5b54e983881)

This will probably require some bot accounts and/or auth tokens.

hzeller, 2023-05-02:
Assigning to Tom to triage.

tgorochowik, 2023-05-16:
I think the easiest way would be to convert https://github.com/chipsalliance/homebrew-verible/blob/835ce1c0f6f1348559d86c211d9ec5b54e983881/Formula/verible.rb to a template (e.g. jinja2 template).

The fields can be filled with the output of:
`git describe` (currently returns `v0.0-3255-gfe86b47d`)
and
`git rev-parse $(git describe)` (currently returns `fe86b47d3fed81d3ba252c1e7c4c7571f65f3fb6`).

The last part would be to configure push access to the `homebrew-verible` repository, we don't necessarily need any dedicated bot accounts, we can configure deploy keys - we can add the private key here as a github secret and then authorize the public key in the `homebrew-verible` repo. This should be sufficient and will probably be the simplest solution as we won't need any extra github accounts.