chipsalliance/verible #1905: Neovim LSP not fully working

adibis, 2023-05-03:
I am trying to get neovim lsp autocomplete working.
I followed the steps as listed in the README file.

I wrote a sample file with a couple of classes. Very simple example is:

`
class aVeryLongName;
endclass : aVeryLongName

class bVeryLongName;
    aVe <<<<------ there is nothing suggested in autocomplete here
endclass : bVeryLongName

`

Trying to figure out how to debug it from here. I also tried adding the project level file and that didn't solve this either.

I am able to see other LSP features such as warnings on bit-width mismatches, trailing whitespaces, etc however, nothing for autocompletion. 

Any idea what might be missing?

hzeller, 2023-05-03:
This is expected, as there are no auto-completion features yet implemented in the language server.

Can you give examples where you'd like to have completion support ? The needs are often different, so I like to understand what are the most common ones. I personally would like to have support for e.g. auto-suggestion of fields in a struct.

aGhandhii, 2023-11-15:
The [Veridian](https://github.com/vivekmalneedi/veridian/tree/master) SystemVerilog Language server implements a completion engine that pulls basic completion options from multiple context sets, found [here.](https://github.com/vivekmalneedi/veridian/blob/master/src/completion/keyword.rs)

Their approach was to break completion requests into the following Contexts:
- Keywords (assert, assign, begin, end, if, else ...)
- System Tasks (error, info, timeformat, random ...)
- Directives (ifndef, endif, include, timescale ...)

When a completion request is triggered by the server, the context is determined (e.g. a '$' character would imply a System Task) and the appropriate pool of keywords is parsed for completion suggestions. The main completion engine can be found [here.](https://github.com/vivekmalneedi/veridian/blob/master/src/completion.rs)

I like this approach as it provides enough filtering to provide relevant completion options without much complexity, and implementing a similar system in Verible would greatly benefit the Language Server.

allRisc, 2024-08-06:
For me the struct capability and class members are the most valuable short term capabilities.