chipsalliance/verible #1910: ReadTheDocs CI throws ImportError with urllib

jbylicki, 2023-05-05:
The docs generation in CI has an issue with the SSL version when trying to use `urllib3`.

The link to an example faulty run: https://readthedocs.org/projects/verible/builds/20425975/

Trace of the Python exception:
```
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/sphinx/registry.py", line 472, in load_extension
    mod = __import__(extname, None, None, ['setup'])
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/sphinx/builders/linkcheck.py", line 19, in <module>
    from requests.exceptions import HTTPError
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/urllib3/__init__.py", line 39, in <module>
    "urllib3 v2.0 only supports OpenSSL 1.1.1+, currently "
ImportError: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with OpenSSL 1.0.2n  7 Dec 2017. See: https://github.com/urllib3/urllib3/issues/2168

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/sphinx/cmd/build.py", line 303, in build_main
    args.tags, args.verbosity, args.jobs, args.keep_going)
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/sphinx/application.py", line 224, in __init__
    self.setup_extension(extension)
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/sphinx/application.py", line 449, in setup_extension
    self.registry.load_extension(self, extname)
  File "/home/docs/checkouts/readthedocs.org/user_builds/verible/envs/1909/lib/python3.7/site-packages/sphinx/registry.py", line 475, in load_extension
    raise ExtensionError(__('Could not import extension %s') % extname, err)
sphinx.errors.ExtensionError: Could not import extension sphinx.builders.linkcheck (exception: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with OpenSSL 1.0.2n  7 Dec 2017. See: https://github.com/urllib3/urllib3/issues/2168)

Extension error:
Could not import extension sphinx.builders.linkcheck (exception: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with OpenSSL 1.0.2n  7 Dec 2017. See: https://github.com/urllib3/urllib3/issues/2168)
```

umarcor, 2023-05-05:
The docs in this repo seem to be markdown sources and markdown files generated in https://github.com/chipsalliance/verible/blob/master/.github/bin/github-pages-setup.sh. That's uploaded to gh-pages: https://github.com/chipsalliance/verible/blob/master/.github/workflows/verible-ci.yml#L575-L599.

ReadTheDocs seems not to be used at all, and `doc/conf.py` does not exist. I wonder how did that ever work.

hzeller, 2023-05-17:
Adding Adam: so if readthedocs is not updated, but the script that should do it fails anyway, can we remove it ?

AdamOlech, 2023-05-19:
I'd say it's safe to remove it. Just like @umarcor pointed out, it never had the chance to work. However, I'm not sure who set it up in the first place (it was not me). It seems like it's something that is done on the admin panel on Read The Docs so removing it doesn't seem to be possible from the `Settings` tab of this repo (i.e. it's not an integration or an OAuth app).

hzeller, 2023-05-19:
It might have been @mithro who set readthedocs up early in the Verible project.