chipsalliance/verible #1916: Use .rules.verible_lint config in VS code

AndreMerendeira, 2023-05-12:
Hello, using VS code v1.78.1 with Verible extension v0.0.3253 (no more SystemVerilog extensions alongside it).
Is there a way to use the .rules.verible_lint config file in VS code?
Adding the path to the file with `verible-verilog-ls --rules_config <path-to-config>` crashes the extension.

AlphonseVA, 2023-05-14:
Have you tried using `verible-verilog-ls ----rules_config_search` ?

AndreMerendeira, 2023-05-15:
@AlphonseVA That also does not work (with and without the extra `--`)

hzeller, 2023-05-17:
@glatosinski can  you help here ?
In general, the rules are looked up [relative to the project root](https://github.com/chipsalliance/verible/tree/master/verilog/tools/ls#adding-project-specific-linter-configuration), but flags should also work (and if not, @glatosinski will make it happen :) ).

closed 2023-05-25 by hzeller