chipsalliance/verible #1918: Verible Language Server not picking up .rules.verible_lint on Sublime Text [Windows][v0.0-3253-gf85c768c]

AlphonseVA, 2023-05-14:
Text Editor: Sublime Text
Version: Build 4143
OS: Windows 11
Active Sublime packages: SystemVerilog  + LSP **only**
Verible: verible-v0.0-3253-gf85c768c-win64

-----

**What activity failed**
`.rules.verible_lint` is defined but the language server still identifies those issues even with the flag `--rules_config_search` being used in the LSP settings command

Screenshot of the text editor:
[image]
See that the LSP Diagnostic Panels are still flagging rules which should have been disabled in `.rules.verible_lint`.

Screenshot of the LSP settings in Sublime Text: 
[image]

I have tried restarting but the rules seem to be still enforced despite being waived in `.rules.verible_lint`. I tried to manually invoke the linting with `verible-verilog-lint --rules_config_search .\ring_ctr.sv` which works as expected. The manual `verible-verilog-lint` successfully picked up `.rules.verible_lint`.

**Expectation**
I would expect the LSP/language server to be able to properly pick up `.rules.verible_lint` and would not flag those warnings that have been waived.

Just some baseless speculations but I am guessing it might be due to how the LSP in Sublime parses the path in Windows. Perhaps it could be missing quotation marks ("") or it's using the wrong backslash or forwardslash symbols for pathing.

AlphonseVA, 2023-05-14:
Logs from troubleshooting verible-verilog-ls on Sublime Text LSP

----

# Troubleshooting: verible-verilog-ls
## Version
 - LSP: 1.23.0
 - Sublime Text: 4143
## Server Test Run
 - exit code: 0
 - output
```
Verible Verilog Language Server built at v0.0-3253-gf85c768c
```
## Server Configuration
 - command
```json
[
  "verible-verilog-ls", 
  "--rules_config_search"
]
```
 - shell command
```sh
verible-verilog-ls --rules_config_search
```
 - selector
```
source.systemverilog
```
 - priority_selector
```
source.systemverilog
```
 - init_options
```json
{}
```
 - settings
```json
{}
```
 - env
```json
{}
```

## Active view
 - File name
```
C:\Users\vincent.tan\Desktop\Testing\ring_ctr.sv
```
 - Settings
```json
{
  "auto_complete_selector": "meta.tag, source - comment - string.quoted.double.block - string.quoted.single.block - string.unquoted.heredoc", 
  "lsp_active": true, 
  "syntax": "Packages/SystemVerilog/SystemVerilog.sublime-syntax"
}
```
 - base scope
```
source.systemverilog
```

## Project / Workspace
 - folders
```json
[
  "C:\\Users\\vincent.tan\\Desktop\\Testing"
]
```
 - is project: True
 - project data:
```json
{
  "folders": [
    {
      "path": "Testing"
    }
  ]
}
```

## LSP configuration

```json
{
  "clients": {
    "verible-verilog-ls": {
      "command": [
        "verible-verilog-ls", 
        "--rules_config_search"
      ], 
      "enabled": true, 
      "selector": "source.systemverilog"
    }
  }
}
```
## System PATH
 - C:\WINDOWS\system32
 - C:\WINDOWS
 - C:\WINDOWS\System32\Wbem
 - C:\WINDOWS\System32\WindowsPowerShell\v1.0\
 - C:\WINDOWS\System32\OpenSSH\
 - C:\Program Files\nodejs\
 - C:\Program Files\Git\cmd
 - C:\Program Files (x86)\Common Files\EMC\
 - C:\Users\vincent.tan\AppData\Local\Programs\Python\Python311\Scripts\
 - C:\Users\vincent.tan\AppData\Local\Programs\Python\Python311\
 - C:\Users\vincent.tan\AppData\Local\Microsoft\WindowsApps
 - C:\Users\vincent.tan\AppData\Local\Programs\Microsoft VS Code\bin
 - C:\Users\vincent.tan\AppData\Roaming\npm
 - C:\Users\vincent.tan\Downloads\verible-v0.0-3253-gf85c768c-win64
 -

hzeller, 2023-05-17:
@glatosinski can you help ?

glatosinski, 2023-06-16:
This issue should be fixed now since https://github.com/chipsalliance/verible/pull/1941

AlphonseVA, 2023-06-16:
Hiya,

Can confirm that this issue has been resolved as of the latest version of the Windows binary.

Verible: [verible-v0.0-3313-gddcea377-win64
 
Screenshot as proof that it's working:
[image]

Rules specified in `.rules.verible_lint` as been successfully waived away from the language server.

Thanks,
AlphonseVA

closed 2023-06-16 by AlphonseVA