chipsalliance/verible #1919: Building from Python

17Reset, 2023-05-15:
Is there a Python binding, please? Considering using verible from Python?

mithro, 2023-05-15:
Surelog might be a better bet -> https://github.com/chipsalliance/Surelog

17Reset, 2023-05-16:
> Surelog might be a better bet -> https://github.com/chipsalliance/Surelog

Thank you very much, does Surelog provide format and obfuscate functions?

hzeller, 2023-05-17:
Surelog mostly provides parsing functionality, and has a Python API to inspect the syntax tree (including elaboration).

Formatting and obfuscating is only provided in Verible.
Verible does not have a public C++ API yet, and thus will also not have a Python API any time soon.
For now it is best to directly invoke the binaries.
If you want, you can use the current C++ libraries, which are very nicely separate, but might be subject to change. A future API will provide a simplified facade around that.

Some tools do have a json output (e.g. the syntax tree output of the syntax-checker; also the linter) to make it easier to interface with scripting language such as JavaScript, Lua, Lisp, Perl or Python.