chipsalliance/verible #1921: Can I only use exe programs?

17Reset, 2023-05-16:
Can the project be compiled as a dll file for developers to use?

hzeller, 2023-05-16:
Internally, the project is organized as libraries that do the tasks for linting, formating etc.
However, there is no stable public API with convenient façade functions around it yet.

What functionality would you like first to be broken out as an API ?

17Reset, 2023-05-17:
> Internally, the project is organized as libraries that do the tasks for linting, formating etc. However, there is no stable public API with convenient façade functions around it yet.
> 
> What functionality would you like first to be broken out as an API ?

Thank you very much, I personally would like the format to be made into a public API first, which will be used very often。