chipsalliance/verible #1923: Change tagging to not include hash ? e.g. v0.0-3255-gfe86b47d -> v0.0.3255

hzeller, 2023-05-16:
Currently, the [tags](https://github.com/chipsalliance/verible/tags) in the releases contain the full output of `git describe`. 

While this is generating unique versions even across branches, that pin-pointing to a particular hash is not needed, but we have not very convenient tags that are hard to easily 'read' as a version number, and in particular impossible to guess to what a 'next' one would be.

So instead of `v0.0-3255-gfe86b47d`, I propose to have the tags numbered as `v0.0.3255`.
It would also be less confusing for packaging folks as they currently seem to come up with different ways to formulate a version number ( seen here: https://repology.org/project/verible/versions ). If we had a common three-dot format of the version number, it would be very unambiguous how to name a package version.

Change would be something like this:
```patch
--- a/.github/workflows/verible-ci.yml
+++ b/.github/workflows/verible-ci.yml
@@ -541,7 +541,7 @@ jobs:
         git config --local user.email "$BOT_EMAIL"
 
         # Create a tag of form v0.0-183-gdf2b162-20191112132344
-        TAG="${TAG:-$(git describe --match=v*)}"
+        TAG="${TAG:-$(git describe --match=v* | sed 's/\(v[0-9]\+\.[0-9]\+\)-\([0-9]\+\)-.*/\1.\2/')}"
 
         git tag "$TAG" || true
         echo "TAG=$TAG" >> $GITHUB_ENV
```

This will result in nice `<major>.<minor>.<patchlevel>` tags such as `v0.0.3255`

Opening this up for discussion, as there might be reasons to _not_ do that, that I am missing ?

Adding @proppy  @mithro @umarcor as folks who have dealt with versioning and packaging before.

umarcor, 2023-05-16:
The `patchlevel` in the proposed format is the number of commits since the last tag. Therefore, it is ambiguous because multiple feature branches with the same number of diverging commits will produce the same version string. As a result, it can only be used if tags are only ever created from a linear main branch (or a linear segment of it) and it is never force-pushed.

hzeller, 2023-05-16:
Yes, but a linear main branch is the only thing we do and release.