chipsalliance/verible #1926: arm64/centos7 fails to build

hzeller, 2023-05-17:
Currently,  the releases are blocked because the [arm64 on centos7](https://github.com/chipsalliance/verible/actions/runs/5006246852/jobs/8971266019) fail to build.

I think if we can show that our static arm64 binary will work on centos7 on arm, we can as well skip remove this build (and possibly others: see #1871 

Assigning to Adam for now as he was working on CI builds recently, but also @umarcor might have some insights here.

hzeller, 2023-05-19:
Maybe just disable that particular build for now.

AdamOlech, 2023-05-19:
Sure, let's do it. I've just created PR: https://github.com/chipsalliance/verible/pull/1928.

AdamOlech, 2023-06-05:
@hzeller just to let you know, we've developed a fix for this issue (https://github.com/antmicro/github-actions-runner-scalerunner/commit/bdec5c447beecda973b09d67af0d820ee010f5b8). Feel free to revert https://github.com/chipsalliance/verible/pull/1928.

hzeller, 2023-06-13:
For distribution, we should consider just distributing static binaries maybe https://github.com/chipsalliance/verible/issues/1871
... but still good to run the build to verify that it would work.