chipsalliance/verible #1929: Rule for undeclared variable

hrivu21, 2023-05-21:
**Summary**

The linter does not report the use of undeclared variables.

**Test cases and examples**
```systemverilog
// This example should be diagnosed.
module ring_ctr #(
    parameter reg[2:0] WIDTH = 4
) (
    input clk,
    output reg [WIDTH-1:0] out
);

    always @(posedge clk) begin
        if (!rstn) out <= 1;
        else begin
            out[WIDTH-1] <= out[0];
            for (int i = 0; i < WIDTH - 1; i = i + 1) begin
                out[i] <= out[i+1];
            end
        end
    end
endmodule
```

**Proposal**

The **rstn** variable is undeclared but used inside the always block. 
I would expect the linter to point this out.

tgorochowik, 2023-05-30:
We should be able to implement this now that we have the symbol table implemented and working

Harleks, 2024-09-07:
> We should be able to implement this now that we have the symbol table implemented and working

Hello, I am trying to implement this type of lint rule using the symbol table.

But I encountered some problems. I cannot access the symble table in the `/analysis/checkers` folder where I am writing the Lint rule.

I have found that symbol tables are currently only used in functions such as “hover” and “goto definition”. None of the lint rules used the symbol table. What should I do if I want to use the symble table in the lint rule in the `/analysis/checkers/` folder?

I want to reference the header file of the symbol table in the lint rule, but I am concerned that referencing the symbol table would go against the original design intention.

fangism, 2024-09-08:
> Hello, I am trying to implement this type of lint rule using the symbol table.
>
> But I encountered some problems. I cannot access the symble table in the `/analysis/checkers` folder where I am [...]

Access might be a matter of bazel visibility.  In this case, it looks like `checkers` should already be allowed to see `symbol-table`, according to https://cs.opensource.google/verible/verible/+/master:verilog/analysis/BUILD;l=5.  Can you share the error message you're seeing?

> I have found that symbol tables are currently only used in functions such as “hover” and “goto definition”. None of the lint rules used the symbol table. What should I do if I want to use the symble table in the lint rule in the `/analysis/checkers/` folder?

At the moment, none of the lint rules use the symbol table, so this feature might be the first such case.  Typically, linter analyses were intended as single-file operations, whereas the symbol table was more intended for multi-file or project-level analysis.  That being said, you can still analyze single files, but expect that external symbol references may be unresolved (and ignore those findings).  

> I want to reference the header file of the symbol table in the lint rule, but I am concerned that referencing the symbol table would go against the original design intention.

Once you're considering referencing information from other files such as headers, you may be better off starting an analysis tool separate from the original single-file linter.  One such place is https://cs.opensource.google/verible/verible/+/master:verilog/tools/project/project_tool.cc.  For instance, one could introduce a new `diagnose` subcommand to report unresolved symbols.  Such a tool could be integrated into editors via the language server protocol.