chipsalliance/verible #1940: Event triggering not recognised as valid syntax

matlupi, 2023-06-05:
**Describe the bug**

There is a syntax error at the -> token.

**To Reproduce**

The minimized test case (test.sv):

```systemverilog
module foo(input clk);
  event a;

  always @(posedge clk)
    -> a;
endmodule
```

Include any options used.

```
>> verible-verilog-syntax ../test.sv
../test.sv:5:5-6: syntax error at token "->"
```

**Actual behavior:**

The valid syntax is rejected

**Expected behavior**

In [SystemVerilog-2017 Standard (LRM)](https://ieeexplore.ieee.org/document/8299595), section 15.5.1.
Syntax 15-1

```verilog
event_trigger ::= 
 -> hierarchical_event_identifier ;
| ->> [ delay_or_event_control ] hierarchical_event_identifier ;
```

Lightborne, 2023-07-17:
Hello,
I am new to how verible development works - I see tgorochowik added this to the release roadmap above.  What does that mean in terms of an ETA for a release?

Also, are there any known workarounds?

tgorochowik, 2023-08-03:
Hi @Lightborne it most definitely is! I cannot provide any ETA at this moment though. Would you be interested in looking into this yourself and submitting a PR?

corco, 2023-08-09:
@Lightborne I don't have the issue when decorating the always block in a begin/end. Could that be a workaround?

I have very few event in my codebase, if you can confirm it only fails on naked always block, it would help narrow the issue.

The following two cases passes:

```systemverilog
module foo (
    input clk
);
  event a;

  always @(posedge clk) begin
    ->a;
  end
endmodule
```

```systemverilog
module foo(input clk, b);
  event a;

  always @(posedge clk)
    if (b)
      -> a;
endmodule
```

Lightborne, 2023-08-14:
@corco  - wrapping in begin/end is a good workaround.  Thanks!