chipsalliance/verible #1952: how to parser a file with macro？

ddppt-yy, 2023-07-03:
imput file like this：

`module A (
`ifdef DEBUG
input debug_tag,
`endif
input xxx,
);
endmodule`

but verible-verilog-syntax cant parse debug_tag, unless add 
`define DEBUG`.

how to add "define DEBUG" in cmd line?

ddppt-yy, 2023-07-03:
I think my issue like this [#1898](https://github.com/chipsalliance/verible/pull/1898)

fpgauserdude, 2023-08-08:
This is really too bad. Verilator was doing exactly what I needed, but the preprocessor is sort of half-way implemented. Example: 

```
ifdef TEST1
foo test1
  (.clk (port.clk));
endif

ifndef TEST2
foo test2
  (.clk (port.clk));
endif
```
But unfortunately, since there is no way to control in the "verible-verilog-syntax" command what preprocessor defines are present, "test2" will be present in the tree and "test1" will not. I was prepared to use the "rawtokens" output to essentially do my own preprocessing. But I can't because not enough information is put into the tree.

Any suggestions?