chipsalliance/verible #1956: LSP not working on macos

lisphacker, 2023-07-09:
A couple of questions first

 - IDE: VSCode 1.80.0 on macos 13.4.1 (verible installed with brew)
 - What other SystemVerilog plugins are active alongside ? None
-----

**What activity failed**
All activity in general. It seems like the lsp binary was detected since it shows the right version number on the Output log, but it does not seem to be doing anything else.

**Expectation**
Syntax highlighting, hover preview, go to def, ...

**What actually happened**
Nothing

**Test case**

```systemverilog
module adder (
    input a,
    input b,
    output s,
    output c
);
    assign {c, s} = a + b;
endmodule
```

**Logfiles**

```txt
Verible Verilog Language Server built at v0.0-3181-g5a92de1f
I0709 02:50:49.287971       1 symbol-table-handler.cc:64] Found file list under /Users/gautham/work/learn/verilog/adder/verible.filelist
I0709 02:50:49.324230       1 json-rpc-dispatcher.cc:66] Ignoring notification 'initialized'
```

[image]

hzeller, 2023-07-10:
Syntax highlighting will not be provided by the language server (at least not yet; syntax highlight has been recently added to the language server protocol, so we might consider that though it requires some chatty protocol; in the meantime, having also a plug-in that does syntax highlighting such as the `Verilog-HDL/SystemVerilog/Bluespec` plugin helps).

The go to definition should work. e.g. if you go in your example on the 'b' symbol and go-to-def, it should jump to the `b` in the input port definition.

For useful things to show in hover, we have #1922 as work in progress.

If you want to see more detailed information in the log output and maybe spot issues that go wrong there, you can set some environment variables to get more detailed log messages; e.g. invoke it like so:

```
VERIBLE_LOGTHRESHOLD=0 VERIBLE_VLOG_DETAIL=1 code adder.v
```

lisphacker, 2023-07-11:
Thanks! The go-to-def still wasn't working, but those env variables showed that verible.filelist was missing. After adding that, go-to-def started working.

I'll try that extension for syntax highlght/hover.

closed 2023-07-11 by lisphacker