chipsalliance/verible #1962: Build fails on undefined `uint32_t` in protobuf

nathanbeckmann, 2023-07-13:
Here is what I get after doing a `bazel clean --expunge`:

```
➜  verible git:(master) bazel build -c opt //...
INFO: Analyzed 1898 targets (0 packages loaded, 0 targets configured).
INFO: Found 1898 targets...
ERROR: /home/beckmann/.cache/bazel/_bazel_beckmann/d932d629abbf7de2514b0932259df566/external/com_google_protobuf/src/google/protobuf/compiler/BUILD.bazel:73:11: Compiling src/google/protobuf/compiler/subprocess.cc failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 63 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/compiler/_virtual_includes/command_line_interface/google/protobuf/compiler/subprocess.h:47,
                 from external/com_google_protobuf/src/google/protobuf/compiler/subprocess.cc:33:
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:186:67: error: 'uint32_t' was not declared in this scope
  186 |     : absl::disjunction<std::is_same<T, int32_t>, std::is_same<T, uint32_t>,
      |                                                                   ^~~~~~~~
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:56:1: note: 'uint32_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   55 | #include "google/protobuf/port_def.inc"
  +++ |+#include <cstdint>
   56 | 
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:186:75: error: template argument 2 is invalid
  186 |     : absl::disjunction<std::is_same<T, int32_t>, std::is_same<T, uint32_t>,
      |                                                                           ^
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:187:67: error: 'uint64_t' was not declared in this scope
  187 |                         std::is_same<T, int64_t>, std::is_same<T, uint64_t>,
      |                                                                   ^~~~~~~~
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:187:67: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:187:75: error: template argument 2 is invalid
  187 |                         std::is_same<T, int64_t>, std::is_same<T, uint64_t>,
      |                                                                           ^
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:188:45: error: template argument 2 is invalid
  188 |                         std::is_same<T, bool>> {};
      |                                             ^~
bazel-out/k8-opt-exec-2B5CBBC6/bin/external/com_google_protobuf/src/google/protobuf/_virtual_includes/protobuf_lite/google/protobuf/port.h:188:45: error: template argument 4 is invalid
ERROR: /home/beckmann/.cache/bazel/_bazel_beckmann/d932d629abbf7de2514b0932259df566/external/com_google_protobuf/src/google/protobuf/BUILD.bazel:142:11 Middleman _middlemen/@com_Ugoogle_Uprotobuf_S_Ssrc_Sgoogle_Sprotobuf_Cwkt_Ucc_Uproto-BazelCppSemantics_build_arch_k8-opt-exec-2B5CBBC6 failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 63 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
INFO: Elapsed time: 2.004s, Critical Path: 1.84s
INFO: 67 processes: 57 internal, 10 linux-sandbox.
FAILED: Build did NOT complete successfully
```

hzeller, 2023-07-14:
interesting, this looks like `cstdint` include is missing on the protobuf dependency we pull in.
I'll update the protobuf dependency, then let's see.

For reference, what is the system environment you're building on ? Linux/BSD distribution (typically `cat /etc/lsb-release` on Linux), gcc version (`gcc -v`) ?

hzeller, 2023-07-14:
Can you try again with latest head ? The protobuf version we used was indeed missing the cstdint header (and now I am wondering how that ever was working, probably some accidental includes that your set-up was not doing)

nathanbeckmann, 2023-07-15:
The `uint32_t` error is gone, thanks, but now I am seeing this error:

```
➜  verible git:(master) bazel build -c opt //...
INFO: Analyzed 1898 targets (0 packages loaded, 0 targets configured).
INFO: Found 1898 targets...
ERROR: /home/beckmann/.cache/bazel/_bazel_beckmann/d932d629abbf7de2514b0932259df566/external/com_google_boringssl/BUILD:152:11: Compiling src/crypto/bytestring/cbb.c failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 36 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In function 'CBB_add_asn1',
    inlined from 'CBB_add_asn1_octet_string' at external/com_google_boringssl/src/crypto/bytestring/cbb.c:567:8:
external/com_google_boringssl/src/crypto/bytestring/cbb.c:388:14: error: storing the address of local variable 'child' in '*cbb.child' [-Werror=dangling-pointer=]
  388 |   cbb->child = out_contents;
      |   ~~~~~~~~~~~^~~~~~~~~~~~~~
external/com_google_boringssl/src/crypto/bytestring/cbb.c: In function 'CBB_add_asn1_octet_string':
external/com_google_boringssl/src/crypto/bytestring/cbb.c:566:7: note: 'child' declared here
  566 |   CBB child;
      |       ^~~~~
external/com_google_boringssl/src/crypto/bytestring/cbb.c:565:36: note: 'cbb' declared here
  565 | int CBB_add_asn1_octet_string(CBB *cbb, const uint8_t *data, size_t data_len) {
      |                               ~~~~~^~~
In function 'CBB_add_asn1',
    inlined from 'CBB_add_asn1_bool' at external/com_google_boringssl/src/crypto/bytestring/cbb.c:578:8:
external/com_google_boringssl/src/crypto/bytestring/cbb.c:388:14: error: storing the address of local variable 'child' in '*cbb.child' [-Werror=dangling-pointer=]
  388 |   cbb->child = out_contents;
      |   ~~~~~~~~~~~^~~~~~~~~~~~~~
external/com_google_boringssl/src/crypto/bytestring/cbb.c: In function 'CBB_add_asn1_bool':
external/com_google_boringssl/src/crypto/bytestring/cbb.c:577:7: note: 'child' declared here
  577 |   CBB child;
      |       ^~~~~
external/com_google_boringssl/src/crypto/bytestring/cbb.c:576:28: note: 'cbb' declared here
  576 | int CBB_add_asn1_bool(CBB *cbb, int value) {
      |                       ~~~~~^~~
cc1: all warnings being treated as errors
INFO: Elapsed time: 3.053s, Critical Path: 2.81s
INFO: 81 processes: 51 internal, 30 linux-sandbox.
FAILED: Build did NOT complete successfully
```

This is strange because I have installed verible from source several times on this machine recently, so I don't know why I'm running into these issues now.

And I'm on OpenSUSE:

```
➜  verible git:(master) cat /etc/os-release 
NAME="openSUSE Tumbleweed"
# VERSION="20230616"
```

kbieganski, 2023-07-17:
I have the same error on Arch with GCC 13.1.1.
Clang 15.0.7 builds it successfully.

hzeller, 2023-07-17:
Interesting, in the CI we explicitly compile with an ancient compiler (gcc 10) to see that we don't use too new features, and a newer compiler for the c++23 compatibility test ... but that is also only gcc 12.
So maybe time to include gcc 13.

I'll reproduce this on my local machine and then see if updating boringssl will fix this.

hzeller, 2023-07-17:
I could not reproduce that with gcc 13.1.0, which is the latest on my distro, but it looks like it is some static analysis finding.
Just updated boringssl, can you try again ?

nathanbeckmann, 2023-07-17:
That fixed it, thanks!

(Why does `verible` pull in boringssl anyway?)

hzeller, 2023-07-17:
We only need it for some sha256 hashing in Kythe. Slightly heavy dependency for that :/

hzeller, 2023-07-18:
@ivan444 do you know if we even need sha256 there, or just some sufficiently unique content hash ?
Possible solutions
  * We could just assemble some unique enough content hash with `std::hash` (maybe concatenate some of the 64Bit values with different seeds)
   * Do a simple sha256 implementation straight from [RFC6234](https://www.rfc-editor.org/rfc/rfc6234.txt) and put it in third_party/
   * Leave it as-is and take the hit of downloading a roughly 50MiB (compressed!) external dependency and compile it even though we only need a tiny fraction.

PR #1982 (2023-07-26): Remove boringssl dependency. Replace sha256 with absl::Hash.
Closes #1962
closed 2023-07-26 by hzeller