chipsalliance/verible #1974: how compile verible on windows？

durongze, 2023-07-21:
Administrator@MY-PHONE MINGW64 /d/code/verilog/verible
# bazel build -c opt --config=create_static_linked_executables //...
WARNING: Ignoring JAVA_HOME, because it must point to a JDK, not a JRE.
Loading:
Loading:
Loading: 0 packages loaded
Loading: 0 packages loaded
    currently loading:  ... (5 packages)
Loading: 0 packages loaded
    currently loading:  ... (5 packages)
INFO: Repository rules_python instantiated at:
  D:/code/verilog/verible/WORKSPACE:152:14: in <toplevel>
  E:/msys64/home/administrator/_bazel_administrator/rs3thaef/external/com_google_protobuf/protobuf_deps.bzl:101:24: in protobuf_deps
  E:/msys64/home/administrator/_bazel_administrator/rs3thaef/external/com_google_protobuf/protobuf_deps.bzl:20:17: in _github_archive
Repository rule http_archive defined at:
  E:/msys64/home/administrator/_bazel_administrator/rs3thaef/external/bazel_tools/tools/build_defs/repo/http.bzl:372:31: in <toplevel>
INFO: Repository rules_license instantiated at:
  D:/code/verilog/verible/WORKSPACE:6:13: in <toplevel>
Repository rule http_archive defined at:
  E:/msys64/home/administrator/_bazel_administrator/rs3thaef/external/bazel_tools/tools/build_defs/repo/http.bzl:372:31: in <toplevel>
INFO: Repository rules_cc instantiated at:
  D:/code/verilog/verible/WORKSPACE:67:13: in <toplevel>
Repository rule http_archive defined at:
  E:/msys64/home/administrator/_bazel_administrator/rs3thaef/external/bazel_tools/tools/build_defs/repo/http.bzl:372:31: in <toplevel>
WARNING: Target pattern parsing failed.
ERROR: Skipping '/...': error loading package under directory '': error loading package 'third_party/absl/absl/utility': Label '//absl:copts/configure_copts.bzl' is invalid because 'absl' is not a package; perhaps you meant to put the colon here: '//:absl/copts/configure_copts.bzl'?
ERROR: error loading package under directory '': error loading package 'third_party/absl/absl/utility': Label '//absl:copts/configure_copts.bzl' is invalid because 'absl' is not a package; perhaps you meant to put the colon here: '//:absl/copts/configure_copts.bzl'?
INFO: Elapsed time: 15.220s
INFO: 0 processes.
ERROR: Build did NOT complete successfully
Loading: 0 packages loaded
    currently loading:  ... (5 packages)

hzeller, 2023-07-22:
What version of bazel are you using (`bazel --version`) ?

The CI only [needs to download llvm](https://github.com/chipsalliance/verible/blob/edfca850d58e4301edd55ebf05b52e44af151054/.github/workflows/verible-ci.yml#L502-L512) for things to work.

You also need `bash`, but the error message does not look like it would run into such trouble.

Here is a comment that contains some [information that might be needed on the Windows environment.]( https://github.com/chipsalliance/verible/issues/1669#issuecomment-1415497724)

I'd first suggest to get the regular compile working (without `create_static_linked_executables`) as static building has not tested on Windows yet, so this eliminates one potential issue.

durongze, 2023-07-29:
D:\code\verilog\verible>set BAZEL_SH="C:\Program Files\Git\bin\bash.exe"

D:\code\verilog\verible>set BAZEL_VC="E:\Program Files\Microsoft Visual Studio\2022\Enterprise\VC"

D:\code\verilog\verible>set BAZEL_VC_FULL_VERSION=14.36.32532

D:\code\verilog\verible>set BAZEL_WINSDK_FULL_VERSION=10.0.20348.0

D:\code\verilog\verible>set BAZEL_LLVM="E:\Program Files\Microsoft Visual Studio\2022\Enterprise\VC\Tools\Llvm\x64"

D:\code\verilog\verible>echo JAVA_HOME=C:\Program Files\Java\jre-1.8\
JAVA_HOME=C:\Program Files\Java\jre-1.8\

D:\code\verilog\verible>java -version
java version "1.8.0_371"
Java(TM) SE Runtime Environment (build 1.8.0_371-b11)
Java HotSpot(TM) 64-Bit Server VM (build 25.371-b11, mixed mode)

D:\code\verilog\verible>echo PYTHONHOME=E:\program\python\
PYTHONHOME=E:\program\python\

D:\code\verilog\verible>python --version
Python 3.7.4

D:\code\verilog\verible>bazel --version
bazel 7.0.0-pre.20230530.3

D:\code\verilog\verible>bazel build -c opt --enable_runfiles //...
WARNING: Ignoring JAVA_HOME, because it must point to a JDK, not a JRE.
ERROR: C:/users/administrator/_bazel_administrator/rs3thaef/external/local_config_sh/BUILD:6:14: syntax error at 'C': expected ,
INFO: Repository win_flex_bison instantiated at:
  D:/code/verilog/verible/WORKSPACE:103:19: in <toplevel>
  D:/code/verilog/verible/bazel/win_flex_bison.bzl:110:26: in win_flex_configure
Repository rule remote_win_flex_bison defined at:
  D:/code/verilog/verible/bazel/win_flex_bison.bzl:100:40: in <toplevel>
INFO: Repository rules_ruby instantiated at:
  D:/code/verilog/verible/WORKSPACE:152:14: in <toplevel>
  C:/users/administrator/_bazel_administrator/rs3thaef/external/com_google_protobuf/protobuf_deps.bzl:109:24: in protobuf_deps
  C:/users/administrator/_bazel_administrator/rs3thaef/external/com_google_protobuf/protobuf_deps.bzl:20:17: in _github_archive
Repository rule http_archive defined at:
  C:/users/administrator/_bazel_administrator/rs3thaef/external/bazel_tools/tools/build_defs/repo/http.bzl:372:31: in <toplevel>
INFO: repository @win_flex_bison' used the following cache hits instead of downloading the corresponding file.
 * Hash '8d324b62be33604b2c45ad1dd34ab93d722534448f55a16ca7292de32b6ac135' for https://github.com/lexxmark/winflexbison/releases/download/v2.5.25/win_flex_bison-2.5.25.zip
If the definition of 'repository @win_flex_bison' was updated, verify that the hashes were also updated.
INFO: repository @rules_ruby' used the following cache hits instead of downloading the corresponding file.
 * Hash 'c88dd69eb50fcfd7fbc5d7db79adc6631ef0e1d80b3c94efe33ac5ee3ccc37f7' for https://github.com/protocolbuffers/rules_ruby/archive/5cf6ff74161d7f985b9bf86bb3c5fb16cef6337b.zip
If the definition of 'repository @rules_ruby' was updated, verify that the hashes were also updated.
ERROR: D:/code/verilog/verible/bazel/BUILD:14:11: While resolving toolchains for target //bazel:flex: invalid registered toolchain '@local_config_sh//:local_sh_toolchain': no such target '@local_config_sh//:local_sh_toolchain': target 'local_sh_toolchain' not declared in package '' defined by C:/users/administrator/_bazel_administrator/rs3thaef/external/local_config_sh/BUILD (Tip: use `query "@local_config_sh//:*"` to see all the targets in that package)
ERROR: Analysis of target '//bazel:flex' failed; build aborted
INFO: Elapsed time: 0.549s
INFO: 0 processes.
ERROR: Build did NOT complete successfully
Loading: 13 packages loaded
    currently loading: @com_google_protobuf// ... (12 packages)
    Fetching repository @rules_pkg; starting
    Fetching ...sers/administrator/_bazel_administrator/rs3thaef/external/rules_pkg; Extracting rules_pkg-0.7.0.tar.gz
    Fetching ...inistrator/_bazel_administrator/rs3thaef/external/win_flex_bison; Extracting win_flex_bison-2.5.25.zip
    Fetching ...el_administrator/rs3thaef/external/rules_ruby; Extracting 5cf6ff74161d7f985b9bf86bb3c5fb16cef6337b.zip

durongze, 2023-07-29:
How to compile verible offline on Windows？

hzeller, 2023-07-29:
You are using a very new bazel and apparently that is not working on your machine yet (see the error at local_config_sh/BUILD:6:).

I suggest to use an older version such (starting from bazel 4 is supported) that is working on your Windows machine.
I think in the CI we use bazel 5.