chipsalliance/verible #1979: Alignment issues when declaring multiple signals of the same type on the same line

XDH-233, 2023-07-23:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
  input logic signed [W - 1 : 0] a, b,
  output logic signed [W : 0] y
```

Include any options or configuration used.

**Actual output**

```systemverilog
// This doesn't look right.
    input  logic signed [W - 1 : 0] a,
    b,
    output logic signed [    W : 0] y
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
    input  logic signed [W - 1 : 0] a,b,
    output logic signed [    W : 0] y
```

Citations to published style guides would help.

zhutmost, 2023-07-24:
maybe this can be also acceptable?
```SystemVerilog
    input  logic signed [W - 1 : 0] a,
                                    b,
    output logic signed [    W : 0] y
```

This style is commonly used be CIRCT/Chisel-generated code.

xpww, 2023-08-03:
I also met this problem, it's giving me a headache, do you have any solution?

vzionv, 2024-03-04:
The same issue for this problem. I think the better way is to automatically infer and complete the missing parts to achieve alignment.