chipsalliance/verible #1985: Obfuscator doesn't work as expected

IEncinas10, 2023-07-29:
Obfuscator fails to obfuscate in certain situations. 

**To Reproduce**

Given the following snippet:

```sv
module moduleName #(
    parameter int WORKS = `WORKS,
    parameter int DOESNTWORK = `DOESNTWORK
) ();
endmodule
```

`verible-verilog-obfuscate < bug.sv`

**Actual behavior:**

```sv
module lnwn10Fdmo #(
    parameter int mGPxd = `mGPxd,
    parameter int wrxQq8xPPJ = `DOESNTWORK
) ();

endmodule
```

**Expected behavior**

```sv
module lnwn10Fdmo #(
    parameter int mGPxd = `mGPxd,
    parameter int wrxQq8xPPJ = `wrxQq8xPPJ
) ();

endmodule
```

**Details**

`verilog-verible-syntax --printtree bug.sv`

Shows different interpretation of \`WORKS and \`DOESNTWORK

```
Node @2 (tag: kTrailingAssign) {
  Leaf @0 (#'=' @45-46: "=")
  Node @1 (tag: kExpression) {
    Node @0 (tag: kFunctionCall) {
      Node @0 (tag: kReference) {
        Node @0 (tag: kLocalRoot) {
          Node @0 (tag: kUnqualifiedId) {
            Leaf @0 (#MacroIdentifier @47-53: "`WORKS")
...
Node @2 (tag: kTrailingAssign) {
  Leaf @0 (#'=' @84-85: "=")
  Node @1 (tag: kExpression) {
    Node @0 (tag: kMacroGenericItem) {
      Leaf @0 (#MacroIdItem @86-97: "`DOESNTWORK")
```

Which I'm not sure if its intended or not, but makes the last macro go unfiltered as it doesn't go into this case:

https://github.com/chipsalliance/verible/blob/a1cd07b13572f19df5776381eb52287e994df392/verilog/transform/obfuscate.cc#L72

But ends up in 
https://github.com/chipsalliance/verible/blob/a1cd07b13572f19df5776381eb52287e994df392/verilog/transform/obfuscate.cc#L83

**Question**

I'm unable to see what is going wrong here. If the code is correctly parsed and the fix just involves adding more cases to the switch in [obfuscate.cc](https://github.com/chipsalliance/verible/blob/a1cd07b13572f19df5776381eb52287e994df392/verilog/transform/obfuscate.cc) I would be willing to submit a PR.

Thanks!

hzeller, 2023-07-31:
Probably you just have to add the `MacroIdItem` to that switch/case.

`MacroIdentifier` and `MacroIdItem` are different tokens in the lexer depending on if they are at the end of the line. I suspect this case as simply be forgotten in that case.

CC @fangism Is my assumption correct ?

fangism, 2023-07-31:
Correct, `MacroIdentifier` and `MacroIdItem` only differ based on position at end-of-line.
Adding `MacroIdItem` should fix your test case.  
Thanks for reporting this bug!

PR #1986 (2023-08-01): obfuscator: add missing case in ObfuscateVerilogCodeInternal
Fixes #1985. Added a simple test case along with the fix.
commit 17c596d3 (2023-08-01): obfuscator: add missing case in ObfuscateVerilogCodeInternal

fixes #1985
closed 2023-08-01 by hzeller