chipsalliance/verible #1987: Unable to parse Content-Length when present in header

aGhandhii, 2023-08-01:
Using `vim 9.0.1403` on Windows 10

Verible Release: `v0.0-3389-ga1cd07b1`

I am using a [vim9 LSP Client](https://github.com/yegappan/lsp), the functionality and configuration is similar to that of [vim-lsp](https://github.com/prabirshrestha/vim-lsp)

My configuration (in `.vimrc`):

```
call LspAddServer([#{
    \   name: 'verible-verilog-ls',
    \   filetype: ['systemverilog', 'verilog'],
    \   path: 'verible-verilog-ls',
    \   args: ['--column_limit=80', '--indentation_spaces=4', '--rules=+line-length=length:80'],
    \ }])
```

-----

_When opening a *.sv or *.v file, the Language Server fails to start with the following error message:_
```
Verible Verilog Language Server built at v0.0-3389-ga1cd07b1
No `Content-Length:` header. 'Content-Length: 2780
Content-Type: application/vim-jsonrpc; charset=utf-8

{"method":"initialize","jsonrpc":"2.0","id":1,"params":{"rootUri":"file:///C:/Users/../../../Verilog/","initializationOptions":{},"workspaceFolders":[{"uri":"file://...'
Statistics
Largest message seen: 0 kiB 
```

_This is confusing because the header sent to the Language Server - as shown in the error message - contains the `Content-Length` parameter as expected._

-----

It seems that parsing is controlled in [message-stream-splitter.cc](https://github.com/chipsalliance/verible/blob/ca53c5d4fcc9eb3430e5e74420a0b191639770a8/common/lsp/message-stream-splitter.cc#L50), but I am unsure what else to do to diagnose this issue.

hzeller, 2023-08-01:
Turns out, the header-parsing was a bit simplistic and didn't expect a header to follow `Content-Length`. Can you try wit the latest release ?

aGhandhii, 2023-08-01:
It works! Thanks for the help

closed 2023-08-01 by aGhandhii