chipsalliance/verible #1993: build fails on windows

fpgauserdude, 2023-08-09:
I have spent quite a long time trying to get verible to build on windows.

I have looked at all the similar issues I can find. I suspect there is something different with my environment from what other people are doing. I cannot get bazel to work without installing Visual Studio 2019. Here are my build steps:

1. install LLVM (choco install llvm); this installs clang version 16
2. set the environment for Bazel: set BAZEL_LLVM, set BAZEL_VC
3. attempt to build (bazel build -c opt --concompatible_enable_cc_toolchain_resultion //...)

I get the following error after about half of the project builds:
$ bazel build -c opt --incompatible_enable_cc_toolchain_resolution //...
Starting local Bazel server and connecting to it...
INFO: Analyzed 1903 targets (133 packages loaded, 5149 targets configured).
INFO: Found 1903 targets...
ERROR: D:/work/sandbox/verible/verilog/analysis/checkers/BUILD:856:11: Compiling verilog/analysis/checkers/v2001_generate_begin_rule.cc [for tool] failed: (Exit 2): cl.exe failed: error executing command (from target //verilog/analysis/checkers:v2001-generate-begin-rule) C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64\cl.exe /nologo /DCOMPILER_MSVC /DNOMINMAX /D_WIN32_WINNT=0x0601 /D_CRT_SECURE_NO_DEPRECATE ... (remaining 28 arguments skipped)
verilog/analysis/checkers/v2001_generate_begin_rule.cc(47): error C7555: use of designated initializers requires at least '/std:c++20'
INFO: Elapsed time: 64.565s, Critical Path: 26.30s
INFO: 566 processes: 68 internal, 498 local.
FAILED: Build did NOT complete successfully

I have tried running using the "/std:c++20" option. That strangely makes no difference, even though bazel fills the screen with warnings about how the C++ standard default option of /17 is being overridden with /20. 

But again, I strongly suspect there is something different about my environment since other people seem to be getting bazel to work without installing visual studio. What am I missing?

I want to echo others suggestions that bazel might be more trouble than it is worth. But I'd really like to adopt verible.

corco, 2023-08-09:
Bazel is using MSVC instead of clang-cl (which Verible does not support).

It should pick clang-cl as a compiler from the options in .bazelrc. On a "standard" setup, you shouldn't need BAZEL_LLVM, BAZEL_VC or `--incompatible_enable_cc_toolchain_resolution`

What does `bazel build -c opt //...` outputs?

fpgauserdude, 2023-08-09:
Yeah, I suspected something like that. But when I attempt to run bazel without those options, it immediately fails with the following error:

$ bazel build -c opt //...
INFO: Analyzed 1903 targets (133 packages loaded, 5162 targets configured).
INFO: Found 1903 targets...
ERROR: D:/work/sandbox/verible/common/analysis/BUILD:15:11: Compiling common/analysis/citation.cc failed: (Exit 1): vc_installation_error_x64.bat failed: error executing command (from target //common/analysis:citation) external\local_config_cc\vc_installation_error_x64.bat /nologo /DCOMPILER_MSVC /DNOMINMAX /D_WIN32_WINNT=0x0601 /D_CRT_SECURE_NO_DEPRECATE /D_CRT_SECURE_NO_WARNINGS /bigobj /Zm500 /EHsc /wd4351 /wd4291 ... (remaining 23 arguments skipped)

The target you are compiling requires Visual C++ build tools.
Bazel couldn't find a valid Visual C++ build tools installation on your machine.

Visual C++ build tools seems to be installed at C:\Program Files\Microsoft Visual Studio\2022\Professional\VC
But Bazel can't find the following tools:
    VCVARSALL.BAT, cl.exe, link.exe, lib.exe, ml64.exe
for x64 target architecture

Please check your installation following https://bazel.build/docs/windows#using

corco, 2023-08-09:
Which version of bazel are you using?

The latest CI used bazel 6.3.1 with LLVM 15.0.7. It shouldn't hurt to pass the bazel option `--compiler=clang-cl` also, although it should be picked up by .bazelrc automatically

fpgauserdude, 2023-08-09:
Should I try updating bazel?

$ bazel --version
bazel 6.2.1

$ bazel build -c opt --compiler=clang-cl //...
INFO: Analyzed 1903 targets (0 packages loaded, 0 targets configured).
INFO: Found 1903 targets...
ERROR: D:/work/sandbox/verible/common/util/BUILD:212:11: Compiling common/util/spacer.cc failed: (Exit 1): vc_installation_error_x64.bat failed: error executing command (from target //common/util:spacer) external\local_config_cc\vc_installation_error_x64.bat /nologo /DCOMPILER_MSVC /DNOMINMAX /D_WIN32_WINNT=0x0601 /D_CRT_SECURE_NO_DEPRECATE /D_CRT_SECURE_NO_WARNINGS /bigobj /Zm500 /EHsc /wd4351 /wd4291 ... (remaining 21 arguments skipped)

The target you are compiling requires Visual C++ build tools.
Bazel couldn't find a valid Visual C++ build tools installation on your machine.

Visual C++ build tools seems to be installed at C:\Program Files\Microsoft Visual Studio\2022\Professional\VC
But Bazel can't find the following tools:
    VCVARSALL.BAT, cl.exe, link.exe, lib.exe, ml64.exe
for x64 target architecture

Please check your installation following https://bazel.build/docs/windows#using

INFO: Elapsed time: 0.896s, Critical Path: 0.37s
INFO: 20 processes: 20 internal.
FAILED: Build did NOT complete successfully

corco, 2023-08-09:
That's progress! This sounds like an issue with abseil or bazel and llvm 16. Can you try llvm 15.0.7 to match our CI run?

It really shouldn't be that finicky. It could be that clang-cl is nowhere as common as MSVC. Back then I didn't support MSVC because designated initializers were not supported before c++20 and abseil didn't support MSVC on c++latest. It would be worthwhile to revisit MSVC so that our toolchain is standard and c++20 support is completed.

fpgauserdude, 2023-08-10:
$ clang-cl --version
clang version 15.0.7
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: C:\Program Files\LLVM\bin

$ bazel build -c opt --compiler=clang-cl //...
INFO: Analyzed 1903 targets (0 packages loaded, 0 targets configured).
INFO: Found 1903 targets...
ERROR: C:/users/<path>/_bazel_<path>/gigcqpgm/external/com_google_absl/absl/log/internal/BUILD.bazel:227:11: Compiling absl/log/internal/nullguard.cc failed: (Exit 1): clang-cl.exe failed: error executing command (from target @com_google_absl//absl/log/internal:nullguard) C:\Program Files\LLVM\bin\clang-cl.exe /nologo /DCOMPILER_MSVC /DNOMINMAX /D_WIN32_WINNT=0x0601 /D_CRT_SECURE_NO_DEPRECATE /D_CRT_SECURE_NO_WARNINGS /bigobj /Zm500 /EHsc /wd4351 /wd4291 /wd4250 /wd4996 ... (remaining 25 arguments skipped)
In file included from external/com_google_absl/absl/log/internal/nullguard.cc:15:
In file included from external/com_google_absl\absl/log/internal/nullguard.h:27:
In file included from C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.37.32822\include\array:9:
C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.37.32822\include\yvals_core.h(847,1): error: static assertion failed: Error in C++ Standard Library usage.
_EMIT_STL_ERROR(STL1000, "Unexpected compiler version, expected Clang 16.0.0 or newer.");
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.37.32822\include\yvals_core.h(502,5): note: expanded from macro '_EMIT_STL_ERROR'
    static_assert(false, "Error in C++ Standard Library usage.")
    ^             ~~~~~
1 error generated.
INFO: Elapsed time: 0.758s, Critical Path: 0.36s
INFO: 16 processes: 16 internal.
FAILED: Build did NOT complete successfully

corco, 2023-08-10:
It seems latest MSVC (which clang-cl use for stdc++ implementation) [only supports Clang-16](https://github.com/microsoft/STL/commit/a62109595b6d89e08172fdf4beb75a2670fe0cc9), and abseil fails on clang-16.

Our [CI uses MSVC 2019](https://github.com/actions/runner-images/blob/main/images/win/Windows2019-Readme.md), is it possible to use that instead of the latest 2022? I am not a Windows user myself, so I am not sure how you would normally deal with version issues like this one. The lack of support for clang 15 must be recent since Github's Windows 2022 image uses clang-15 with MSVC 2022.

You can also try to use clang-16 and update absl in the file ` WORKSPACE` to see if a newer version works. If that is the case, I would encourage you to submit a PR.

I will try to find a computer to work on this, but I don't have a Windows one on hand so it will wait a little while.

fpgauserdude, 2023-08-10:
The way to force MSVC 2019 is to use the environmental variable that I mentioned when I opened this issue. I did that, and I got farther. Here is the new error:

ERROR: D:/path/verible/verilog/tools/lint/BUILD:181:8: Executing genrule //verilog/tools/lint:line-length-in-module-body-waived failed: (Exit 1): bash.exe failed: error executing command (from target //verilog/tools/lint:line-length-in-module-body-waived) C:\Windows\system32\bash.exe -c ... (remaining 1 argument skipped)
sed: -e expression #1, char 1: unknown command: `|'
INFO: Elapsed time: 8.162s, Critical Path: 4.46s
INFO: 3232 processes: 3126 internal, 106 local.
FAILED: Build did NOT complete successfully

corco, 2023-08-10:
Is that a WSL2 bash? This is not supported by bazel as far as I remember.

The bash should be a msys2 bash like the one from git bash. Our CI has (from Windows2019 image) Git bash, Msys2 bash and WSL2 bash, I did my development with Git bash.

fpgauserdude, 2023-08-10:
Well...I was able to compile successfully in cygwin. The windows bash.exe error was suspicious.

I am not sure how to control the selection of various tools in bazel. Should I have been running from a GIT bash prompt? I was just using a vanilla windows command prompt.

At any rate, cygwin works. Maybe I'll try a few other things too. 

Side note...I strongly suspect that Cmake would work better than bazel for windows builds, and would make life a lot easier for people like me who really need windows builds.

Thanks for your help!