chipsalliance/verible #2002: Logic declarations not formatted like ports , even with same align settings

MahmoudKMaarouf, 2023-08-21:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
```
[image]

I expect it to look like this:

[image]

Because there is a return between the declarations, I do not expect it to take the spacing from the previous declarations.
The ports behave exactly as I want logic declarations to behave. Here is how the output looks:

[image]

kbrunham-intel, 2026-09-22:
@hzeller This issue still does not have a reduced, copy-pasteable `.sv` reproducer that we can run with `verible-verilog-format`. Could the original reporter (or anyone hitting this) please attach a minimal snippet? Without that it is hard to confirm whether the bug is still present on v0.0-4294-gc1d8f5e8 or to add a regression test.

MahmoudKMaarouf, 2026-09-22:
Hi @kbrunham-intel it's not that hard to reproduce. I've provided the text below. 

logic [SEG6-1:0] single_l_f;
logic [SEG6-1:0] single_r_f;
logic [SEG6-1:0] single_0_f;
logic [SEG6-1:0] single_1_f;

logic [SEG1-1:0][WIDTH -1:0] multi_l_a [NUM_X-1:0];
logic [SEG1-1:0][WIDTH -1:0] multi_r_a [NUM_X-1:0];
logic [SEG1-1:0][WIDTH -1:0] multi_0_a [NUM_X-1:0];
logic [SEG1-1:0][WIDTH -1:0] multi_1_a [NUM_X-1:0];