chipsalliance/verible #2003: Is there a way to make the right parenthesis aligned in instantiations ?

MahmoudKMaarouf, 2023-08-21:
Is there a way to make the right parenthesis aligned in instantiations ?

**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
```

Formatter output:
[image]

Expectation/ desired output:
[image]

kulya97, 2024-01-26:
是的

kbrunham-intel, 2026-09-22:
@hzeller This issue still does not have a reduced, copy-pasteable `.sv` reproducer that we can run with `verible-verilog-format`. Could the original reporter (or anyone hitting this) please attach a minimal snippet? Without that it is hard to confirm whether the bug is still present on v0.0-4294-gc1d8f5e8 or to add a regression test.

MahmoudKMaarouf, 2026-09-22:
dut_dummy #(
    .TEST_PARAM1          (TEST_PARAM1),
    .TEST_PARAM2          (TEST_PARAM2),
    .NUM_ERRORS           (NUM_ERRORS),
    .NUM_ERRORS_TO_STICKY (NUM_ERRORS_TO_STICKY),
    .RAM_SIZE             (RAM_SIZE)
) u_dut_dummy (
    .clk    (clk),
    .rst    (rst),
    .a      (a),
    .b      (b),
    .i_err1 (i_err1),
    .c      (c),
    .idk    (idk),
    .dego   (dego),
    .dego2  (dego2),
    .dego3  (dego3),
    .dego4  (dego4)
);