chipsalliance/verible #2019: Language server crashing when requesting definition of a symbol

drgarbi, 2023-09-28:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ?
   - VS Code
 - IDE version:
   - 1.82.2
 - What other SystemVerilog plugins are active alongside ?
   - [Verilog-HDL/SystemVerilog/Bluespec SystemVerilog](https://marketplace.visualstudio.com/items?itemName=mshr-h.VerilogHDL)
 - Verible extension version: v0.0.3416
-----

**What activity failed**
go-to-definition, and hover
Everytime I have my cursor over at any symbol in the code and I press control, the LSP crashes.
This can happen multiple times and it quickly reaches the 5 crashes threshold that disable the LSP restart.

**Expectation**
Control over symbol enables go-to-definition and hover information

**What actually happened**
LSP crashes

**Test case**

If needed, include SystemVerilog sample file(s) that show the code you want to do an operation on
```systemverilog
// Sample SystemVerilog file in case
```

**Logfiles**

Here are the logfiles:

```txt
Set language server executable (verible-verilog-ls) doesn't exist or cannot be accessed
Using executable from path: /home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/bin/verible-verilog-ls
Verible Verilog Language Server built at v0.0-3416-g470e0b95
F0928 08:49:22.710157 1665833 symbol_table.cc:2021] Check failed: node->Parent() == nullptr 
*** Check failure stack trace: ***
    @           0x60eb8d  absl::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x407997  verilog::ResolveReferenceComponentNode()
    @           0x48cb50  verible::ApplyPreOrder<>()
    @           0x49458a  verilog::DependentReferences::Resolve()
    @           0x494632  std::_Function_handler<>::_M_invoke()
    @           0x48d08c  verible::MapTree<>::ApplyPreOrder()
    @           0x494712  verilog::SymbolTable::Resolve()
    @           0x4852b8  verilog::SymbolTableHandler::BuildProjectSymbolTable()
    @           0x485f84  verilog::SymbolTableHandler::Prepare()
    @           0x48601b  verilog::SymbolTableHandler::FindDefinitionLocation()
    @           0x43cf4b  std::_Function_handler<>::_M_invoke()
    @           0x608a00  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x60904d  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x609483  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x609a61  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x609b4d  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x4324ee  verilog::VerilogLanguageServer::Run()
    @           0x4297cc  main
    @           0x6b6eca  __libc_start_call_main
*** SIGABRT received at time=1695901762 on cpu 5 ***
PC: @           0x6ef01c  (unknown)  __pthread_kill
    @           0x6ced70  (unknown)  (unknown)
[Info  - 8:49:22 AM] Connection to server got closed. Server will restart.
[Error - 8:49:22 AM] Request textDocument/definition failed.
Error: Connection got disposed.
	at Object.dispose (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:197385)
	at Object.dispose (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:277471)
	at _.handleConnectionClosed (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:277684)
	at _.handleConnectionClosed (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:338037)
	at t (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:275773)
	at i.invoke (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:198998)
	at o.fire (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:199759)
	at Y (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:186643)
	at i.invoke (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:198998)
	at o.fire (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:199759)
	at m.fireClose (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:207640)
	at Socket.<anonymous> (/home/dgarbi/.vscode-server/extensions/chipsalliance.verible-0.0.3416/dist/extension.js:2:209225)
	at Socket.emit (node:events:525:35)
	at Pipe.<anonymous> (node:net:322:12)
Verible Verilog Language Server built at v0.0-3416-g470e0b95

```

I could not set the VERIBLE_LOGTHRESHOLD to 0 since I am running it over an SSH connection. If additional logs are needed, any guidance on that would be appreciated.

hzeller, 2023-10-24:
This should be fixed by https://github.com/chipsalliance/verible/pull/2029 - can you try again ?

drgarbi, 2023-11-01:
Hello, @hzeller.

I have tried again using the latest release binary found at: https://github.com/chipsalliance/verible/releases/download/v0.0-3428-gcfcbb82b/verible-v0.0-3428-gcfcbb82b-linux-static-x86_64.tar.gz

It has been fixed now! Thank you

closed 2023-11-01 by drgarbi